samphonic/Moppy2 #87: Clicking and not moving

GlorifyedToasterWithLegs, 2020-02-09:
Hey! I'm not entirely sure if this is a wiring issue or just a problem with the motor, but whenever I play a song, all the motor does is make a very quiet clicking sound, but the head remains still. I'm just wondering if you have dealt with this before.

SammyIAm, 2020-02-09:
Just had the same thing come up in #85.  Try switching the step and direction wires.

GlorifyedToasterWithLegs, 2020-02-09:
alright, so when I do switch them, the drive just makes less clicking noises, so I'm thinking that the drive is definitely trying to move as the clicks are to the beat of the song from what I can tell, but I'm still not sure of how to go about this.

GlorifyedToasterWithLegs, 2020-02-10:
ok so update, I switched my power supply (which was just a 5v plug soldered onto a power connecter) to an actual computer power supply. This has solved the problem, but that still begs the question, why does my 5v plug not work for powering the drive?

SammyIAm, 2020-02-10:
Glad you've got it working.  When you were using the 5v supply, did you
connect the ground from the supply (either directly or via the floppy
drive) to the GND pin on the Arduino?

It sounds like there may have been a grounding issue.  (It's maybe also
possible that the "5v" supply you were using was outputting something
different enough from the Arduino's 5v that communications were affected,
but that seems less likely than the grounding thing)

closed 2020-08-12 by Sammy1Am