samphonic/Moppy2 #90: Device mapping according to note velocity

ComputerSongs, 2020-02-25:
Hi. I would like to know if you can map notes of different velocities, to play in a floppy tower, just as the Lothean instrument does:
https://github.com/Lothean/Moppy2/tree/master/Arduino/Moppy/src/MoppyInstruments
but mapped from the same MoppyControlGUI, and using only FloppyDrives.cpp, since I am having problems with more than 6 floppy drives with FloppyTower.cpp.
The problem happens after resetting the floppy drives, when the STARTUP_SOUND is going to sound, the first one just sounds, and the arduino does not connect to the PC.
I think it happens because there is too much information to process for the Arduino, and I suppose it would be better for the java program to do it.
Thank you.

SammyIAm, 2020-02-27:
Had to doublecheck the code, but yes, that is currently possible in the ControlGUI (but it's a little unwieldy), basically:

- In the mapping section of the UI you'll need to create as many mappings as you have drives.  Then, in the condition script, add `&& v>0` to the first drive, `&& v>20` to the next, &c..  This will cause the first drive to play all notes, the second to only play when the velocity is above 20, and so on. 
- You'll also need to change the SubAddress script to be a hard-coded value (`1` for the first drive, `2` for the second, &c.)
- And finally, since you're essentially playing one channel of information on all these drives, the condition script should hard-code the same channel for all of these (e.g. your first two conditions would be `c==1 && v>0` and `c==1 && v>20`

The "correct" way to do this is probably on the Arduino-side.  It shouldn't really be too much additional information to be processed, though I see that @Lothean 's code also has a decay added to the velocity that might be making things a little more complicated.

Hopefully that's useful information though.  I'm intrigued by what this would sound like so I'd love to see a video if you get a it working!

ComputerSongs, 2020-02-27:
Thanks, but I had some inconvenience. First, only the first floppy disk drive was turned off at the time of the NOTEOFF, solved it by adding a conditional to all other floppy disk drives in `void tick ()` to stop sounding when the first one turned off:
  `if (currentPeriod[2]>0)`
. This worked but I have one more problem, in ControlGUI configure all the mapper:

[image]

(the first mapper is just a 5¼ disk drive that sounds on channel 1, the tower works on channel 2)

But the tower simply sounds all the notes on all floppy drives regardless of their velocity.
Maybe velocity is not giving the correct values. By the way, I do not know if it affects but I do not use midi files, but I use MIDI-IN with virtual MIDI cable directly to the editor, and from the editor I am sending notes of different velocity.

SammyIAm, 2020-02-27:
Ah, yeah, I hadn't thought about note-off events (another reason this should probably all be done on the Arduino-side rather than in the GUI, but it sounds like you found a workaround).

As for the mapping, it looks like things should be fine (and thank you for the screenshot, that really helps!).  As far as I know, MIDI-IN should be more or less identical to using a MIDI file (both just feed MIDI events into Moppy).

The value of `v` is being set [on this line here](https://github.com/SammyIAm/Moppy2/blob/21216a5e3673ac08d527844e993caffebecb4473/Java/MoppyLib/src/main/java/com/moppy/core/events/mapper/MIDIScriptMapper.java#L66).  Since all the drives are playing, it would appear that the velocity is always being set to 127.  Some things to maybe try:

- Try switching one of the mappers to e.g. `c==1 && v<=48` and see if that particular drive stops playing (just to make sure the Condition script in the mapper is working as expected).
- If that does work, you could also try setting the condition to `c==1 && v==127` just to confirm the velocity is *always* being seen as 127.
- You might give a MIDI file a try just on the off-chance that there *is* a difference from MIDI-IN.
- If you're willing to try modifying the Java code, you can try adding `System.out.println(midiMessage.getData2())` (you might need to format the data value to get it to print out something readable) after the line linked above.  This should at least tell you what value is being seen as the velocity, though if the first things work, you can be pretty sure you know what it is.

Let me know how that goes.  If the mappers are buggy or the velocity is coming back less than 127, then there's some code that needs fixing.  If the mappers work and the velocity is always 127, I suspect either something with the virtual MIDI-IN, or your editor.

ComputerSongs, 2020-02-27:
It's weird, now that you eliminate the condition `if (currentPeriod[2]>0)` it already responds correctly to the velocity issue but, as soon as the note ends, the others do not stop sounding, they only change the note, until I send him a RESET or a PAUSE. 
It does the same with both MIDI file and MIDI-IN.
This means that the problem is not in ControlGUI, but that it is in the Arduino code?

Oh by the way, the conditions in the mapper are only for NOTE-ON events, so a MIDI NOTE-OFF event may send a `v = 0`, and the NOTE-OFF data , will not be sent, since it does not meet the condition to pass.
It's just a hypothesis, I don't program java but I think it can be that, since it works the same way to send NOTE-OFF to their respective subAddress
Perhaps it is better for the Arduino to do it, just to solve the problem of having more than 6 drives.

ComputerSongs, 2020-02-28:
Bah, it works, just add a condition for the NOTE-OFF to pass when `v==0`
`c==1 && (v>=48 || v==0)`

SammyIAm, 2020-02-28:
Ah, nice work, glad you got it working!  When you said, "the conditions in the mapper are only for NOTE-ON events", I thought, "Wait a minute, the conditions should apply for *both* events."  So that ended up being the key.

closed 2020-08-12 by Sammy1Am