samphonic/Moppy2 #92: Problem with netbeans

PablooAjofrin, 2020-03-08:
I cannot start using moppy because netbeans doesn´t load it, I always press on Import project from Zip but I always receive a message saying No netbeans project added, I´ve downloaded it from the releases part. Can anyone help me?

SammyIAm, 2020-03-09:
If you want to load the project in NetBeans, you'll want to download the source code by using the green "Clone or download" button on the project's main page.  The releases contain just the parts needed to run the program on its own.

PablooAjofrin, 2020-03-09:
> If you want to load the project in NetBeans, you'll want to download the source code by using the green "Clone or download" button on the project's main page. The releases contain just the parts needed to run the program on its own.

I solved it by using the moppy gui and installing java 8, because when I tryied to use it, it just opened the cmd, blink and close,  Now I'm able to reproduce sound in my floppy disk but it is really low and it barely moves. Do u know why?

SammyIAm, 2020-03-09:
You'll need to be a bit more descriptive (or link to a video).  If by "really low" you mean the *volume* is low, that's probably correct.  "Quiet" operation would be a "feature" of normal floppy drives, so they're not designed to be noisy.

The read head *should* be moving from one side to the other of its track as in other videos.  Is it moving less than that?

PablooAjofrin, 2020-03-09:
> You'll need to be a bit more descriptive (or link to a video). If by "really low" you mean the _volume_ is low, that's probably correct. "Quiet" operation would be a "feature" of normal floppy [...]

yeah, it doesn't move, it just sounds but the volume is really really low and I can't aprecciate if it is generating some musical notes or not.

SammyIAm, 2020-03-09:
It definitely should be moving.  

- Is it just doing a sort of ticking thing?  If so, you might try swapping your STEP and DIRECTION wires, they might be backward.
- What MIDI file are you using to test?
- When you reset or power on the Arduino, does the first drive play a few notes?

PablooAjofrin, 2020-03-17:
I finally got it, thanks a bunch!

closed 2020-03-17 by PablooAjofrin