samphonic/Moppy2 #94: 17 Drives

C288, 2020-03-30:
Hi, it has been a while since I have messed with this project but I am at it again. I figured out how to add a hard drive. (You just have to turn a hard drive into a hard drive speaker). I have 16 floppy drives working but I need to get a 17th slot for the HDD (It gets connected the same way a floppy drive does). I don't understand what I have to add to the FloppyDrives.cpp. I also don't understand what all the "LOW"s are on line 39. Mainly how many do I need to add?

SammyIAm, 2020-03-31:
Hello!

Unfortunately 16 is a bit of a magical number for MIDI systems as it's the number of MIDI channels that exist.  This means you can't easily assign notes to a 17th device in the MIDI file, so you'll need to come up with some clever mapping scripts to get separate notes, or just have it play the same notes as one of the drives.

You didn't ask about MIDI files though, so I'm thinking you've made it past that part.  Here's a (partial, maybe) list of changes you'll need to make:

- `MoppyConfig.h` needs to be updated to reflect the new maximum number of sub-addresses (even though MIDI is limited to 16, Moppy supports up to 255 of these).
- All the arrays on lines 31-49 will need an extra spot added for the HDD.
- The `currentState` array needs **two** spots added.
- The `tick()` function needs a block added for `currentPeriod[17]` to toggle the HDD pins.

And that should about do it.  The comment for `currentState` explains:

> Array to keep track of state of each pin.  Even indexes track the control-pins for toggle purposes.  Odd indexes track direction-pins.  LOW = forward, HIGH=reverse

`LOW` and `HIGH` are just `#defines` in the Arduino library that resolve to `0` and `1` respectively (and ground-state or 5-volts in the physical world), but are used here for clarity to signify that they are pin-states (rather than actual integers).  These all need to be tracked because the code needs to "toggle" pins sometimes requiring knowledge of their current state.

Hope that's helpful!

C288, 2020-03-31:
I actually did not realize that about the midi channels. How would I be able to manually assign a specific channel to it?

C288, 2020-03-31:
@SammyIAm I just thought of this too. If I remember correctly channel 10 is mostly for drum events. Would there be a way to tell the arduino to play channel 10 through drive 17? And duplicate another channel to another drive?

SammyIAm, 2020-03-31:
If you wanted channel 10 to be assigned to sub-address 17, you'd put `c==10` as the condition script (so that mapper would only apply to notes on channel 10), and you'd put `17` as the sub-address script (so all those notes would be assigned to sub-address 17.  If you wanted all the other channels to work as normal, you could add a second mapper with `c!=10` (so the mapper applies to everything *except* 10), and `c+1` as the sub-address (so that each channel gets its own sub-address).

The sub-address refers to the drive connected to the Arduino.  `c` refers to the channel (but starts at 0 and goes to 15).  So `c+1` is basically just assigning each channel to its own sub-address.

If you wanted the extra drive to duplicate a channel (e.g. channel 3), you could do a third mapper with `c==3` and a sub-address of `10` (since drive #10 won't be getting any events because you mapped them over to 17).

C288, 2020-03-31:
Thank you so much! You are always super helpful.

closed 2020-08-12 by Sammy1Am