samphonic/Moppy2 #95: error compiling

Gasper6201, 2020-04-01:
hi, i recently aquired a nucleo board so i thought i'd finally try this project and i'm running into this issue. i am completely new to this so bare with me.
[error.txt](https://github.com/SammyIAm/Moppy2/files/4412069/error.txt)

SammyIAm, 2020-04-01:
While I'm happy to **bear*** with you, this doesn't seem to be an error with this project per se.

What program are you using to compile the sketch?  There's a lot of errors about missing types; have you imported the Arduino framework for whatever board you're using?

-- Actually, now that I'm looking closer, most of these issues seem to be stemming from code that's not in this repository.  Have you made modifications to the code yourself, or did you download this code from somewhere else?

Gasper6201, 2020-04-04:
I have downloaded the code directly from this site the adapted one for stm32, I am working with arduino ide and downloaded everything to program the nucleo board I have. I will try to post the files I have as soon as I get to my computer.

Gasper6201, 2020-04-04:
https://github.com/SammyIAm/Moppy2/pull/21

This is the link. If it's not under this forum I am sorry, I'm quite lost on this site.

SammyIAm, 2020-04-08:
That's an old pull request that was never merged (from a repo that doesn't seem to exist any more).  Unfortunately since I don't have those boards, I can't really help you out too much.  From the gist of things though, it seems like the bits that need to be changed to work on an STM32 board are:

- Pin mappings
- Timer

You've got two options: either start from the master branch in this repository and try to make the adjustments yourself, or try to fix the code you have now.  The former will require a bit more knowledge, but might end up being cleaner for you in the end.

The latter should start in two places:

- `uint8` isn't a known type for some reason.  You'll need to either include whatever's needed make that work, or change it to something that is known.
- There's some timer weirdness.  There's a decent chance it's just a library mismatch, since `HardwareTimer` seems to exist, but be expecting different arguments.  Maybe try editing the timer parts to match the documentation of the latest STM32 libraries.

Wish I could give you something more definitive, sorry.

closed 2020-08-12 by Sammy1Am