samphonic/Moppy2 #96: An error occurred when running bat

wudixiaobinbin, 2020-04-26:
Exception in thread "main" java.lang.NoSuchMethodError: com.fazecast.jSerialComm.SerialPort.setBaudRate(I)V
        at com.moppy.core.comms.bridge.BridgeSerial.<init>(BridgeSerial.java:30)
        at com.moppy.control.NetworkManager.lambda$new$0(NetworkManager.java:55)
        at java.util.ArrayList.forEach(ArrayList.java:1257)
        at com.moppy.control.NetworkManager.<init>(NetworkManager.java:54)
        at com.moppy.control.MoppyControlGUI.main(MoppyControlGUI.java:41)
Am I miss any lib?

SammyIAm, 2020-04-26:
Hmm, that's a new one.

If you got your download from [the releases page](https://github.com/SammyIAm/Moppy2/releases) then everything should have been included.  Maybe try re-downloading it?

If you're not using the releases download (i.e. if you're trying to build and/or run from the source), make sure you're using Maven to build the project (since it will download the libraries for you).

wudixiaobinbin, 2020-04-27:
I tried to re-downloaded it, but still that error.
I also tried 2.0.0 and 2.0.1, they all same error.

SammyIAm, 2020-04-27:
Huh, weird.  I guess next questions are:

- What version of Windows are you running?
- What version of Java are you running?

It looks like the version of jSerialComm that I'm using for Moppy (2.0.2) is a bit behind the latest (2.6.1).  If you're willing to download the source and build it, you might try updating the [build.gradle file](https://github.com/SammyIAm/Moppy2/blob/master/Java/MoppyLib/build.gradle) for MoppyLib to reference the latest version and see if that helps.

wudixiaobinbin, 2020-04-28:
My Windows version is Win10 17134 Enterprise
My Java version is jdk1.8_251

I'll try to build the source, but I'm a PHP developer, I may be can't figure it out.

SammyIAm, 2020-04-29:
Those versions seem like they should be fine.  I've built a version with the newer jSerialComm:

[MoppyControlGUI-2.0.2.zip](https://github.com/SammyIAm/Moppy2/files/4549202/MoppyControlGUI-2.0.2.zip)

Give that a try and see if anything changes.

wudixiaobinbin, 2020-04-29:
Yes, the 2.0.2 just worked fine, thank you.

SammyIAm, 2020-04-30:
Glad that worked, and thanks for letting me know.  Must be some sort of issue with the older version of jSerialComm; I'll have to make sure I push that change to the next release.

wudixiaobinbin, 2020-05-01:
There is a new problem, The 2.0.2 MoppyControlGUI can't work with 2.0.1 Arduino code

SammyIAm, 2020-05-01:
Well, the easiest thing would be to use the latest official 2.0.1 release.

However, because I'm always curious about improving things: how can you
tell that it doesn't work?  Are you getting an error, or have you found a
piece of code that looks to be an issue?

SammyIAm, 2020-05-01:
Apologies for that previous response, I got this thread mixed up with another one. 😳

The reason the build I gave you isn't working is because I built it off the experimental branch rather than mainline (because it's what I had on hand, and I didn't know it wouldn't be backward compatible).  So there's a few options:

- I can try to push out a release in the next few days, but you might have to wait a bit.  (I need to do this anyway, but I'd like to get a few other changes in and make sure I can test it first)
- You could try building it yourself with the updated gradle.build files (for the latest jSerialComm).
- Depending on what sort of incompatibility you're running into, I might be able to walk you through a quick fix in the Arduino code.  What issue are you running into?

wudixiaobinbin, 2020-05-01:
I tried to build the 2.0.1 myself, and there is a weird phenomenon that I can running it with Eclipse but can't running with NetBeans. They are using same JDK, and they all can't build a jar, just can running with Eclipse.
The Arduino code could work well with 2.0.1 which I running with Eclipse.
And my Arduino is Mega 2560 with 16 floppy drivers. 2.0.2 couldn't work with my Arduino may because I modified the Arduino code to compatible 16 floppy drivers.

SammyIAm, 2020-05-01:
Huh, weird.  It's been a long time since I actually tried NetBeans, so maybe something has changed.  Glad you've at least got it working in Eclipse to test.

What happens when you try to run the 2.0.2 Java program with the 2.0.1 Arduino code?

wudixiaobinbin, 2020-05-01:
1. After choose a COM port, There are no info appear on the right.
2. When playing mid music, Mega's TX light is flashing but floppy drivers doesn't have any sounds.
3. In Serial Monitor, I can see the program sends the data, but the max sub-address is 08.

SammyIAm, 2020-05-01:
Ah, okay, I _think_ it's a really easy fix: Just change the baudrate in [https://github.com/SammyIAm/Moppy2/blob/master/Arduino/Moppy/src/MoppyNetworks/MoppySerial.h#L14](MoppySerial.h) to `115200` instead of `57600`.  Then re-upload the Arduino firmware with the change, and I **think** you'll be good to go using the 2.0.2 Java application.

I upped the rate in the experimental branch (and therefore the 2.0.2 build I attached above) to see how it worked, and it seems to work fine.

wudixiaobinbin, 2020-05-02:
Thank you, it's worked fine now

Sammy1Am, 2020-08-12:
Fixed in newer commits.

closed 2020-08-12 by Sammy1Am