samphonic/Moppy2 #97: Can't run ControlGUI bat when uploaded or connect floppy/arduino to NetBeans

Cobaltra, 2020-04-30:
Hello, I have a problem trying to get connect my floppy and Arduino to NetBeans and run the program.  My version of NetBeans IDE is 8.0.2 and I was able to upload the ControlGUI to it, but it didn't give me the option to connect a device.  The device drop down bar is clear and doesn't do anything when I click on it.  I used the original controlGUI that came with Moppy2 and I tried 2.0.2 also, but both don't give me an option to connect a device or run the program.  I was able to successfully connect my floppy to the Arduino IDE and it played the jingle.  I power the Arduino from my laptop and the floppies can be powered by an ATX or a 5V barrel plug adapter.  I don't understand what I'm doing wrong and would appreciate some help! Thanks everyone! :)

SammyIAm, 2020-04-30:
I'm not sure what "device drop down bar" you're referring to.  Could you provide a screen shot?

Cobaltra, 2020-04-30:
[image]

Cobaltra, 2020-04-30:
I am nowhere familiar with NetBeans or if I'm doing the right steps to connect to it.  The thing I circled is what I believe is the bar to select your serial port, but I may be wrong.  When I click on the "run" tab, everything that tab shows are unclickable.  Is there another IDE I can use that works with Moppy?

Cobaltra, 2020-04-30:
If I try to launch the .bat file of controlGUI from the file explorer, it pulls up the command prompt for a second and then closes.  It does it again, even if you launch it from the command prompt.  I made sure to do my research and saw to unzip the file, as well as unblock it.  It didn't fix it I guess.  I don't know where and how to upload the included songs without running into this problem.

SammyIAm, 2020-04-30:
When you run the .bat file from the command line, what error appears in the
terminal when you run it?  (That'll tell you the issue there)

As for running from NetBeans, you don't need the .bat if you want to
run/debug from NetBeans.  Just have it execute the MoppyControlGUI file
that contains the `main()` method (can't recall at the moment, not at my
computer).

Cobaltra, 2020-05-01:
The problem is that the command prompt box is there for half a second and then it's gone before I can even read what it said.  I tried to screen record it through xbox game bar and it didn't record what the prompt text said.  Where would the 'main() method file be in MoppyControlGUI?  Is it under bin or lib?

SammyIAm, 2020-05-01:
Ah, open the command prompt _first_ (by e.g. typing `cmd` in the location bar when you've opened the `bin` folder) and then type "MoppyControlGUI.bat" to run the .bat file.  This should keep the command prompt open even after the .bat exits.

Cobaltra, 2020-05-01:
[image]
[image]

Cobaltra, 2020-05-01:
I don't understand what that means.

Cobaltra, 2020-05-01:
I typed "cmd" in the "search bin" bar located in the upper right of file explorer when I went into the bin file.  Did I do everything you told me to do correctly?

SammyIAm, 2020-05-01:
You need to be **in** the `bin` directory of the MoppyControlGUI release you downloaded and unzipped.

Cobaltra, 2020-05-01:
Here we go!

Cobaltra, 2020-05-01:
[image]

SammyIAm, 2020-05-01:
And now you've got a [Google-able error](https://www.google.com/search?q=unsupported+major.minor+version+52.0) :)

(You need at least Java 8 installed)

Cobaltra, 2020-05-01:
Okay, I'll go back to 2.0.1 and get Java 8! Thanks for your help so far and I hope this fixes things!

Cobaltra, 2020-05-02:
Oh my gosh, it worked!  Thanks so much, Sam!  I'm now able to open MoppyControlGUI!

Cobaltra, 2020-05-02:
Okay, new problem.  How do I test that I have a dead floppy?  I added my third drive, then added power and the LED doesn't come on.  I made sure that my drive select pins were in the right places and moved them around to 14/15.  I even checked from the opposite side of the pins just to check if this thing was wired backwards.  None of my problem solving techniques worked and I'm stumped.  Any advice?

SammyIAm, 2020-05-08:
It sounds like you've covered most of my initial suggestions (thanks for including what you've already tried).  Only a few other options I can think of:

- It might be worth checking every pair (1&2, 3&4, etc.) to see if _any_ of them get the LED to come on.  (Maybe it's a weird interface)
- With 11/12 connected (and also with 13/14) try manually stepping the drive by: Connecting the `direction` pin (18) to ground (17 works for this), and then tapping the `step` pin (20) to ground (19 would be fine) repeatedly.  If you've got the right pins, the head should step forward once per tap.
- If that works, it's possible the LED is just not working.  I've got a couple drives where the LED never turns on, but they work fine.

closed 2020-08-12 by Sammy1Am