samphonic/Moppy2 #98: Floppy powered but not working

Thunderbird98, 2020-05-22:
Hi. I am new to Github.
I have spent most of the day trying to get this project to work. I am not sure whether the problem lies in the software or hardware.
First I setup the hardware assuming the floppy drive pins are counted left to right. I set up all the software and that seemed fine, but the floppy drive spindle spun constantly and the light and read head did nothing. When I clicked disconnect on moppydesk, the drive would make a small click.  Later, I tried connecting the pins assuming the pins are counted right to left, backwards from the diagram. This time, the drive light came on, but I still got the same click on disconnecting from moppydesk. I also tried switching the step and direction wires to no success. I have tried both configurations on 4 floppy drives: three 3 1/2" drives and one 5 1/4" drive. They all basically do the same thing. What am I doing wrong? The Arduino I'm using is older, but it seems to upload fine. Netbeans seems to work fine too.
Here's a picture of my second configuration:
[image]
Any help would be greatly appreciated!

Sammy1Am, 2020-05-22:
First: Welcome to GitHub, that Arduino should be fine, and thank you for providing a picture, it definitely helps with the troubleshooting.

I think some of the original confusion is that most diagrams are of the male connector on the floppy drive itself, rather than the female connector on the ribbon cable.  I've done a quick diagram of the cable connector below.

[image]

It looks like you're close on your connections, except the STEP and DIRECTION pins are swapped (Arduino 2 -> Floppy 20, Arduino 3->Floppy 18).

When you have it hooked up (after swapping those wires), and press the reset button on the Arduino, does the floppy drive play any startup tones?  Or do anything?

The click when connecting or disconnecting is normal, so you're pretty close.  What MIDI file are you trying to play?

Thunderbird98, 2020-05-22:
Is the spindle supposed to spin at all? I noticed the ribbon cable had a twist in it, but now it is as if I'm connecting backwards like I first did.
I also have two questions about the software (I'm a teenager and am only beginning to learn programming): what is the difference between moppy.ino and TimerOne? They both seem to upload to my Arduino (I have also put the timerone file under the arduino IDE folder). 
My second question is: What is the MoppyControlGUI.bat file for? I click on it and it just opens and immediately closes.
Thanks
[image]
[image]

Thunderbird98, 2020-05-22:
I am trying to play the included tetris.mid file

Sammy1Am, 2020-05-22:
It looks like you've got the cable entirely backward again now.  Pay close attention to the little sticky-outy plastic bit (key) on the cable.

[image]

The drive motor should not be turning on.  If you connect pins 16 or 10 to ground though, those are the motor-enable pins, and would cause that to happen.

TimerOne is a "library" that I'm using to make writing the Moppy code easier.  Specifically, it has a bunch of included functions that make it easy to interact with the hardware timers on the Arduino.  Rather than re-invent the wheel and write all of that from scratch, I'm just importing the TimerOne library for those parts.

`.ino` files are are called "sketches" and unique to the Arduino IDE, but they essentially represent the "code" for the program.  In the case of `Moppy.ino`, it's just an empty placeholder that gets the Arduino IDE to compile the code in the `src` directory (where all the actual code is.  You can even see where I included the TimerOne library [in the FloppyDrives.h](https://github.com/Sammy1Am/Moppy2/blob/master/Arduino/Moppy/src/MoppyInstruments/FloppyDrives.h#L10) file.)

`MoppyControlGUI.bat` run the Java program without needing to use NetBeans.  Closing immediately generally indicates an error.  Assuming you're on Windows, instead of double-clicking, open up the `bin` folder, type `cmd` in the location bar (which should open a command prompt in the bin folder), and then type `MoppyControlGUI.bat` at the command prompt.  If an error occurs, it won't immediately close on you.  Most likely your JAVA_HOME isn't set properly.

Thunderbird98, 2020-05-22:
So pressing reset on the arduino now makes the head move back and forth! Yay! Thats the most progress I've made all day. However, Moppydesk still doesn't do anything when I load the tetris midi. Every time I load Moppydesk on Netbeans, I get this error:

May 22, 2020 3:42:21 PM java.util.prefs.WindowsPreferences <init>
WARNING: Could not open/create prefs root node Software\JavaSoft\Prefs at root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5.
Is that bad?
I also put RXTXcomm.jar in java\jre\lib\ext and rxtxSerial.dll goes in java\jre\bin. Is that right?
Thanks

Sammy1Am, 2020-05-23:
Sorry, got distracted with a phone call there.

MoppyDesk and RXTX are all part of the older (no longer supported) MoppyClassic.  Make sure you're only using the Arduino and Java code from the releases page for Moppy2 ( https://github.com/Sammy1Am/Moppy2/releases ).

Try to get that MoppyControlGUI.bat file working with the command line like I mentioned in my last comment as that's easier than dealing with NetBeans if you're not planning to modify the code.  (Even if you *are* planning to modify the code, it's easier to start that way)

Thunderbird98, 2020-05-23:
I tried doing that in command and it said  java_home was not set
I will do that later. Thank you so much for the help, I will update when I get it working.

Thunderbird98, 2020-05-23:
So I got the floppy drive running with moppy classic, but I cant get 2.0 working. I tried setting java_home to the java folder, but when i run moppycontrolGUI.bat in command, I get this error:

Exception in thread "main" java.lang.UnsupportedClassVersionError: com/moppy/control/MoppyControlGUI : Unsupported major.minor version 52.0
        at java.lang.ClassLoader.defineClass1(Native Method)
        at java.lang.ClassLoader.defineClass(Unknown Source)
        at java.security.SecureClassLoader.defineClass(Unknown Source)
        at java.net.URLClassLoader.defineClass(Unknown Source)
        at java.net.URLClassLoader.access$100(Unknown Source)
        at java.net.URLClassLoader$1.run(Unknown Source)
        at java.net.URLClassLoader$1.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.net.URLClassLoader.findClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.launcher.LauncherHelper.checkAndLoadMain(Unknown Source)
Any help would be greatly appreciated.

Sammy1Am, 2020-05-23:
A quick Google search suggests that this is because you have a version of
Java older than 8 installed:

https://stackoverflow.com/questions/22489398/unsupported-major-minor-version-52-0

Make sure you have at least Java 8.

Thunderbird98, 2020-05-23:
I downloaded Java from this site:
https://java.com/en/download/win10.jsp
and still get the same error

Sammy1Am, 2020-05-23:
- Did you uninstall the previous version of Java?
- Where is JAVA_HOME pointing?

I suspect JAVA_HOME is still pointed at the older version of Java you had installed before.  If you don't need it for anything, it's probably best to just uninstall it.  But either way it sounds like you're still not using Java 8 to launch the MoppyControlGUI.

Thunderbird98, 2020-05-24:
Hi. Sorry I didn't respond yesterday.
My java version is updated to 8, and I thought I had set JAVA_HOME to the correct folder but I still get this:

ERROR: JAVA_HOME is set to an invalid directory: C:\Program Files (x86)\Common Files\Oracle\Java\javapath

Please set the JAVA_HOME variable in your environment to match the
location of your Java installation.

When I type "where java" into command, it says that it's installed in "C:\Program Files (x86)\Common Files\Oracle\Java\javapath", and I set java home using "advanced system settings>environment variables>edit" but command says that that folder is invalid. What am I doing wrong?

Sammy1Am, 2020-05-24:
I can't really help you out as much on this one.  There's plenty general info on Google, but the solution is specific to your setup and where you've installed things.  For example, my `JAVA_HOME` is `C:\openjdk-11` which works for me, but almost certainly wouldn't work for you.

Thunderbird98, 2020-05-25:
I will figure that out myself.
Thank you so much for all your help! Without it I would not be able to get this running.
Before I close this thread, I want to ask one more question: would this work for other devices? I have an internal Zip drive that uses an IDE connector. Is there a configuration for that? I have also seen youtube videos with people using old dot matrix printers as well. Would that work?

Sammy1Am, 2020-05-25:
Good question, but the answer is a solid: maybe. 😉 

The output of the Arduino is basically a square wave at the frequency of the note being played.  Floppy drives have a `STEP` pin, so connecting this to a square wave signal will cause the read head to step at the frequency of the note being played.  In theory, any device that can be made to operate at a given frequency can play music (razors, electric toothbrushes, stepper motors, CNC machines...), but it's a question of how to communicate the frequency from the Arduino to the device.

IDE connections don't have a `STEP` pin, so while the read head in a zip drive, or the print head on a dot matrix printer are capable of stepping like this, the interface isn't as straight forward.

Thunderbird98, 2020-05-25:
I got the moppy 2.0 working!
Thank you for all your help!

Sammy1Am, 2020-05-26:
Sure, glad you got it working!

closed 2020-05-26 by Sammy1Am