samphonic/Moppy2 #100: An inquisition for a Future project.

DJthefirst, 2020-06-07:
I've been following Moppy since Moppy Classic and want to develop a new system based on it. I'm going into college and though I don't know everything about coding I'm planning to learn on the fly throughout my 4 years. the basic design of the project focuses around a central hub controller with 4-16 connections and then modules with 16 channels already set up for different instruments. ex A Floopy Module, ShiftRegister, Stepper motor, Pneumatic controller, Tesla Coil Controller. just to name a few. Software-wise I'm planning to use a system similar to Moppy except each of the 4-16 connections can be configured to run off different midi files. it would be up to the user to make sure the midi files are the same tempo start time etc. But it should allow more than 16 channels for enthusiasts. In software, the user would be able to designate which files are routed to each module through the hub. I'm thinking about using a Poe connection to pass data between modules with a large 12v power bus also connected between. (still tentative) each module should also be able to be connected directly to a computer and run without the hub. I also thought a similar system may be able to be used for moppy between serial ports though it would require a lot of work and I'm not sure of the implications of running multiple midi files in parallel. id imagine you cant run more than 1 instance of a midi natively in windows.  The software and board schematics and files would all be open source but I may sell completed modules and hubs if people are interested. This is very much just an idea I wanted your opinion on as I have limited programming experience and some knowledge of electronics design. I also fully expect this project will take 3-5 years to complete. I have only seen a few projects that have used floppy drives and various motors along with more advanced instruments like pneumatics and tesla coils and those that have open have numerous breadboards strewn about with custom software like some of Flanons work [https://youtu.be/2UesaCcfI_8](https://youtu.be/2UesaCcfI_8) there is possibly a way to do it with multiple midi hubs and software pass through but it's not clean and still consists of many breadboards. Again nothing is finalized and I wanted your opinion and see if you had any interest in such an [endeavor?]

Sammy1Am, 2020-06-07:
Well it certainly sounds like you've got some good ideas, and I like the idea of a central hub with extensible modules, but I do have some thoughts (these are all subjective though, so you're 100% free to disagree of course):

**16+ Instruments**
First of all, while it's _technically_ possible to sync up MIDI file playback across multiple files (you can probably run hundreds or thousands in parallel on a desktop Windows machine), there wouldn't really be any sane way to edit them.  I think you're much better off pursuing a way to handle more than 16 devices from a single file.  Ideally, I think, supporting "tracks" as a delineator is probably the way to go as there's not really an upper bound on the number of tracks in a MIDI file.  You could use "channels" to differentiate between modules, perhaps, and tracks to differentiate between streams of notes (which is more in line with what the MIDI standard intended to use channels for anyway).

The only challenge you'll run into with track delineation is that whatever sequencer library you use will need to provide track information during playback.  Java's included MIDI library *doesn't* do that, which is the only think that's kept me from switching to tracks instead of channels for Moppy.  Depending on the language you use, it might already be included, but even for Java it should be possible to extend the current library.  I've been thinking about trying to do that myself now and again anyway, so I might be able to be convinced to tackle that part to add to Moppy if you're going to use Java 😉 .

**Power/Connections**
When you say "PoE" connection, are you talking about actually using like 802.3 Ethernet for sending data packets, or do you just mean using RJ45 connectors?

You *could* use actual Ethernet for data (you can see this in action with the UDP library being used in the [Moppy experimental branch](https://github.com/Sammy1Am/Moppy2/tree/experimental/).  The advantage here is that you could plug your devices into any standard network (or even use WiFi), but it requires more significant processing power on the module-side of things to decode the information.  Not to mention PoE circuitry to convert the 24-48v down to whatever you need.

If you want to just use serial communication, you could still use RJ45 connectors for quick connect/disconnect, but you may have trouble with longer runs of wire if you try to send power over them.  Especially at 5v (the voltage required for floppy drives), you'll get a fair bit of voltage drop over any sort of length of wire.  If you're supplying 12v through the wires and converting it at the modules, 5v would probably be okay, but you might struggle with 12v devices.

Unless you can be sure all your modules will be using the same voltage, you're probably better off building the hub/data connections separate from the power supply.  Pneumatics and tesla coils especially will need dedicated power supplies anyway.

**Reinventing the wheel?**
Now I want to be clear this isn't motivated by pride, I know Moppy has plenty of limitations, but it's worth examining: why start from scratch instead of using Moppy?  If major changes are needed you could just fork it, but it sounds like it could almost work for you as-is already (16+ devices being the stand out missing feature).

DJthefirst, 2020-06-08:
The plan would indeed be a fork or branch off of Moppy but I wasn't sure how many changes would need to be made to the underlying code to support multiple tracks or files in a new architecture.

### 16+ Instruments
I know that a Windows PC can easily handle running multiple files I have just run into issues while editing midi for my tesla coil where you cannot play a file while it is being edited due to windows media player/encoder being busy. I suppose this may just be from running off the same file but I wasn't sure if you would encounter errors. personally I don't think its that hard to create multiple midi files to allow more instruments as most midi files come with multiple chords per track simply duplicating a file and deleting 1/2 the notes in one file and the other notes in the second file leaves you with 16+ channels with only one note per channel you just repeat until on every single channel there is only 1 note on at any given time. "this is how I currently edit tracks for my tesla coil as it can only play 2 notes at a time except it fits in one midi file as unnecessary instruments like drums are removed". I use Mixcraft but I believe most other programs like Fl studio would allow such manipulation. 

I haven't played around with adding 16 plus tracks to a midi file but it looks like you would have to do a similar thing except converting the midis into a hex or .txt with a program like [](http://flashmusicgames.com/midi/mid2txt.php) splicing and then converting back to midi. I don't see why either method couldn't be used as you just have to strip out the track data one method just requires preprocessing by the user from what I understand.

As for delineating the architecture of the system if we have already stripped the track data from the file, I don't see a reason to even use channels any more except for midi out. In theory, if you have the device id of every module you can just send X number of tracks to the module and each instrument gets a track that you can designate in the GUI and the module should negotiate with moppy how many instruments it can support. I can look into writing a java library but my java experience is limited to FRC robotics and not so much application development and I'm currently finishing up my 5ft DRSSTC (Tesla Coil) so it may be a while before I can get started. 

### Connections...
There are many options for connections but the main 3 I'm looking at is Ethernet to Hub and Then Hub to Modules, Ethernet to a switch to the modules (Basically multiple network connections "Hubless") and, Ethernet to Hub and then a Priportoy/Serial to the devices. Poe was just something to consider if we had a hub that would allow the logic side of the modules to be powered independently of the power side though ethernet switches cost 70$+. a separate most likely 12v power supply would be common between modules allowing the use of a single server PSU and bus to power the setup or the user could use multiple 12v computer power supplies for each module or similar allowing flexibility and a neater setup. for a tesla coil, the module is only the controller as the coil itself is only connected by fiber optic and runs of wall voltage. similar for a flyback transformer or other high voltage device.

**Ethernet to Ethernet With HUB**
-Poe Possibly?
-Adds Complexity and another thing to go wrong
-some more cost
-Simpiflys modules
-Gives more control/features possibly 
-moves processing on many modules to a single hub reducing module cost
-separates logic from instrument drivers 
-each module might be able to be controlled without a hub but with less control

**Ethernet Network**
-Cost$$$
-modules cost more 
-Needs more onboard processing on each module
-each module is controlled directly without a hub full control
-can use current UDP Lib
-cleaner setup 
-more flexible

**Ethernet to Pripotry/serial RJ4/usb**
-possibly confusing to users
-least cost per module and no network card on modules
-possibly use modules independent from the hub with serial
-the least processing on modules as all data handling happens on the hub
-can reuse much of current Arduino code on modules
-Hubs require more processing power basically a mini-computer

Feel free to add any additional pros and cons this is just a quick list

Sammy1Am, 2020-06-08:
### 16+ Instruments
You lost me a little at the "adding 16 plus tracks to a midi file but it looks like you would have to do a similar thing except converting the midis into a hex or .txt with a program" bit.  You wouldn't need to do any preprocessing to support a MIDI file with 16+ tracks, MIDI files already support having tons of tracks.

If you're talking about splitting up chords into multiple streams, then you'd need to do some preprocessing either way, but duplicating each track within a single and file deleting the unneeded notes is the same amount of work, but doesn't require multiple files.

The real issues start to come up if you want to make any changes to the music; e.g. if you wanted to remove a few bars, you'd have to carefully remove the *same* bars across multiple files.  Or if you wanted to change the tempo, you'd have to do it multiple times.  Not to mention the "select file" button starts to get complicated if you have to support an arbitrary number of files.

> There are ways to have multiple instruments on one track...

This is true, but for the purposes of Moppy (and I think your project if I'm understanding correctly), the strategy will likely always be one-track/channel per device, so it's a bit of a moot point.

> As for delineating the architecture of the system if we have already stripped the track data from the file, I don't see a reason to even use channels any more except for midi out.

Moppy is [doing this already actually](https://github.com/Sammy1Am/Moppy2/wiki/Design-Docs) with MoppyMessages.  The messages leaving the Moppy Control application are assigned a Device (or "module") address, and a Subaddress (or each "instrument" on the "module").  Right now this assignment is happening by default based on channel, but anything could be used (like tracks or note range).

### Topology
Now that we're getting into the weeds a bit more, what was your thought on the functionality of the hub?  Each module already needs to be able to process messages into notes for that instrument, there's not really any functionality that I can see required from the hub other than to forward messages from the control application.

DJthefirst, 2020-06-09:
> You lost me a little at this bit. You wouldn't need to do any preprocessing to support a MIDI file with 16+ tracks, MIDI files already support having tons of tracks.

I didn't see a way in the program I use to edit midi files to have multiple tracks playing at the same time on a single channel as once they compiled back into a midi file any overlapping tracks are combined. There is a way to have multiple tracks sequentially with different instruments but that's not breaking the magical 16. It may just be my limited knowledge of the program. The only way I could see working was directly converting a midi file into hex and text splicing them together and then re-exporting back to a single midi. It's not too hard but adds an extra step. I'm generally fine with preprocessing the data by hand to make sure everything comes out right its just the extra step of converting it. I may not be fully understanding which part of a midi file is the track or how you would compose a single midi with 16+ tracks even though the protocol may support it.

> This is true, but for the purposes of Moppy (and I think your project if I'm understanding correctly), the strategy will likely always be one-track/channel per device, so it's a bit of a moot point.

Agreed. It was more of a comparison to how you can have a system automatically separate multiple notes and chords on one track into different tracks compared to doing it by hand which yields higher quality results. I believe Round Robbin works in a similar manner. The goal would also be like you said one track/channel per device/ instrument and multiple instruments per module. To allow control of each instrument individually instead of the current method I've seen used, of sending a single track with multiple notes to a cluster of instruments each instrument being assigned a different note due to the 16 limit.

> Moppy is doing this already actually with MoppyMessages. The messages leaving the Moppy Control application are assigned a Device (or "module") address, and a Subaddress (or each "instrument" on the [...]

I remembered reading this but forgot about it. So the moppy messages support up to 255 devices then and 255 subaddresses for about 65,000 instruments. to send data over ethernet 802.3 do you then link each IP with a device address or send the message to all connected devices similar to a can bus where only one device can talk at a time? I took networking 1 & 2 so I understand the protocols just not all the underlying code of moppy. Also, do you also have a way to negotiate with ethernet connected devices in moppy or would we have to add a protocol? To grab things such as module type number of subaddress devices connected and possibly 1-2 other features or would the user need to set up the number of devices and have no feedback from the device.

> Now that we're getting into the weeds a bit more, what was your thought on the functionality of the hub? Each module already needs to be able to process messages into notes for that instrument, [...]

Honestly, it was just part of the original plan as at the time I needed a way to expand the one ethernet port on a laptop to multiple devices instead of a mess of USB hubs or similar. Though it may still have a few uses and benefits I can come up with a direct connection to the modules from moppy seem like the best option. currently, 

### some pros/uses for a hub would be

- Separating the ethernet connection into multiple serial connections allowing the use of normal Arduino in the system and other serial devices along with cutting some of the cost of modules. It would be a glorified USB hub. (could also be used alongside ethernet modules.)

- In case moppy does not have an interface for collecting data from modules a hub could serve to automatically negotiate and display the state of each device for quick lookup and determine which tracks are sent which devices and channels. and possibly other features I haven't thought of. 

- In case of a display at a fair or maybe in a museum similar to one of MrSolidSnakes setup or eric Goodchilds, a hub could be used in place of a computer. something like a LattePanda or raspberry pie could serve as a cheaper interface for users to interact with. with eventually a playlist system or demo mode, the hub would then run moppy itself.

Sammy1Am, 2020-06-09:
> once they compiled back into a midi file any overlapping tracks are combined

It sounds like whatever program you're using might be saving to MIDI Type 0, which is where all the events are combined into a single track.  MIDI Type 1 files keep all the tracks separate.  As far as I know, Type 1 was way more common than Type 0.  You can see an example in [this file](http://brentworld.tripod.com/knightrider.mid) where there are two tracks assigned to channel 4, two for channel 7, and three for channel 10.  And while it's *possible* to have multiple channels in a single track, it seems like that's generally avoided.

MIDI files don't have any concept of "channels" as a separate thing.  Each note-on and note-off message contain the channel within.  That is, there's no way to specify a note at all without explicitly assigning that particular note a channel.

#### Networking
When using an 802 network (and also more or less similarly when using serial), the messages are sent out using UDP multicast so every device receives every message.  The devices themselves are configured with a Device Address, and only pay attention to messages aimed at that address (or system messages which are interpreted by all consumers).  The sub-address is used to differentiate between instruments connected to the same device.

In the [protocol documentation](https://github.com/Sammy1Am/Moppy2/wiki/Design-Docs) you can see that I have added basic Ping/Pong messages to allow devices to be discovered by the controller (and when you run the GUI, you can see each device and which sub-addresses it supports).  If device-specific configuration was needed, it could be added to this protocol, but so far nothing's really come up that didn't make more sense to keep as a performance-time message.

#### Hub
- Have you considered replacing the Arduinos with ESP8266s?  They've got Wi-Fi, so you could ditch the wires entirely.
- If you're set on using Arduinos, you might have trouble finding a hub that supports enough serial connections, but you could use I2C as discussed in #62 and your hub could just be another Arduino that converts serial messages into I2C messages.
- You could also use a *literal* USB hub, which would act as a serial hub for your computer.

> a hub could be used in place of a computer. something like a LattePanda or raspberry pie could serve as a cheaper interface for users to interact with

LattePandas and Rapberry Pis **are** computers 😉, so that's already an option with the Moppy software right now.  Currently the only controller application requires a GUI, but the functional code is all contained in MoppyLib, so a CLI controller should be relatively trivial.  I know some people have implemented playlist systems in the past (maybe search around GitHub a bit), but I don't remember who off hand.

DJthefirst, 2020-06-09:
Do you know which/any programs that use midi type 1? For ESP8266s a wireless-based system is not ideal as the EMF from a tesla coil would cause massive interference and potentially longer runs depending on the setting ex. if it was done outside with fuller scale actual instruments. Or setting up at a fair for display some have weak connections an setting up a local network is more of a hassle than a simple plug and play Ethernet switch also being cheaper. you can of course always add a wireless system and use ESP8266s but for the backbone and modules, ethernet is a better solution. as for the modules they will not necessarily be Arduinos but an Atmega chip though I'm not sure which with headers for an ethernet module or may be integrated into the board. logic wise. 

The use of a hub as a serial controller was just a side thing to integrate setups that already use an Arduino or users who want to easily and quickly use multiple Arduinos over an internet connection without setting up ethernet modules for each device. I don't really know the application or who would be interested just a quick idea. anatomy wise of a device could be a cheap switch with multiple ethernet modules and small microcontrollers with serial out or a single ethernet module to a larger microcontroller with an ic2 bus to multiple serial or USB outs. again I don't see where this use case for a hub is going just something I came up with quick.

Sammy1Am, 2020-06-10:
From what I can see, Mixcraft should be using Type 1 MIDI files, but I'm also seeing some forum posts about MIDI-weirdness in Mixcraft so 🤷.  I personally use either Sibelius (**super** overkill, but I already knew it when I started), or AnvilStudio.

I think as long as you were careful (which you should be ⚠️⚡ ), you might be okay interference-wise (a quick search turned up [these guys](https://4hv.org/e107_plugins/forum/forum_viewtopic.php?77065) who seem to have been okay with 2.4Ghz signals to control Tesla coils).  And of course the Tesla coils can also cause interference on wired connections.  That said, wired will always be cheaper and more reliable, so I can't argue with that.

Speaking of cost though, Ethernet-capable modules seem like they'll be more pricey than I2C/serial, and a USB/serial connection from the PC to the hub will be cheaper than an Ethernet-powered hub.  Which is not to say using Ethernet isn't technically feasible, but I'm usually in the less-is-more camp for system design.  I'd just say as you think about the use cases for your hub, don't rule out any **features**, but consider the simplest implementation that gets you those features.

DJthefirst, 2020-06-10:
> you might be okay interference-wise (a quick search turned up these guys who seem to have been okay with 2.4Ghz signals to control Tesla coils). 

While it is definitely viable to run a coil off a 2.4Ghz signal the controller is usually a fiber optic connection to the coil. Moppy would be communicating with the coil controller not the coil it’s self which reduces interference. I have no idea the scale of some of the coils talked about but the one picture showed a smaller desktop coil. Some also talked about a trade off between speed and reliability further down in the thread though it doesn’t matter to much as moppy doesn’t require to fast of a connection. 

> Tesla coils can also cause interference on wired connections. That said, wired will always be cheaper and more reliable, so I can't argue with that.

Though wired connections are also susceptible in my experience they are slightly more reliable than wireless and shielded cables nullify most interference if needed. When using udp if a moppy stop frame is dropped to something like a Tesla coil controller module the coil could over current and fail even though their should be built in redundancy’s. Also a possible matter of safety if a something goes wrong and a control frame is dropped. I’d always go for the more reliable option for a little more cost. 

With a wired connection you can also always add An ap to make it wireless while an esp8266 can’t as easily become an Ethernet connection. This is just planning for the modules but in Prototyping we could always test out an wireless module. Users can also still use esp8266 on the network just the modules would be Ethernet standard.

> Speaking of cost though, Ethernet-capable modules seem like they'll be more pricey than I2C/serial, and a USB/serial connection from the PC to the hub will be cheaper than an Ethernet-powered hub

At this point I don’t see a use to using a hub possibly something for far future discussions. The only use I saw was for large stage demos or if someone literally wanted an orchestra where a 50+ ft spi or serial bus connection would be impractical due to signal degradation. (Edge cases)

As for the ethernet connection on the modules a ether net controller ic to serial looks to only be 4-5$ And connectors only 1$. Though we can still use a 15-20$ ethernet module in development I think taking the time to integrate it with the module pcb would be worth it

Sammy1Am, 2020-08-12:
Closing for now due to inactivity, but feel free to reopen if needed.

closed 2020-08-12 by Sammy1Am