samphonic/Moppy2 #99: Drum

odinokoe4mo, 2020-06-07:
How I can include drum part in my floppy player? I have found shift register instruments in your code, but It isn't work. I suspect that problrm in Java moppy controller app, but can you explain me how it works.

WinWins-YT, 2020-06-07:
Ты долбоеб?

Sammy1Am, 2020-06-07:
Well you'll need to be a little more clear about what you're trying to do.  If you're trying to just play a drum part on a floppy drive, the best way I've found is it just us **very** short notes for rhythm (like in [this video](https://www.youtube.com/watch?v=w4gVlhjXWAo)).

If that's not what you're trying to do, what are you trying to use for drums?  What sort of input does the device require?  And when you say the ShiftRegister isn't working, what have you tried so far, what are you expecting, and what are you seeing instead?

odinokoe4mo, 2020-06-08:
I want to play drum part in MIDI file on hard disc. I've tried use ShiftRegister, but it fails.

Sammy1Am, 2020-06-08:
Could you provide a wiring diagram or, even better, pictures of how you've connected the hard drive?  When you say, "it fails," could you be more specific?  What have you tried so far, what are you expecting, and what are you seeing instead?

odinokoe4mo, 2020-06-08:
Can you explain me how works this drum system? Maybe I do something wrong.

Sammy1Am, 2020-06-08:
I really would like to try to help, but you need to answer some of my
questions in order for me to do that.

odinokoe4mo, 2020-06-09:
[image]
This is approximate diagram of my device (because on photo nothing is clear). I connect this, connect arduino with flops (it works perfectly) and load MIDI file with drumline in "Moppy Control" but flops started playing without drums. Individually all components have been checked and work but when I trying to play drum part it fails.

Sammy1Am, 2020-06-10:
Thanks for the description and diagram, that helps quite a bit.  I'm going to have more questions before I have answers, but hopefully we can get this working.

I don't have any experience with using an L298N or connecting a HDD with Moppy, so there's not much I can do to help you out on that side of things.  I will trust that you've connected all of that correctly.

In [MoppyCore](https://github.com/Sammy1Am/Moppy2/blob/master/Arduino/Moppy/MoppyCore.cpp), did you comment out the lines for FloppyDrives and uncomment the lines for ShiftRegister?

Also in MoppyCore, there are [a few lines](https://github.com/Sammy1Am/Moppy2/blob/master/Arduino/Moppy/src/MoppyInstruments/ShiftRegister.cpp#L20-L22) that limit which MIDI notes will be played.  The ShiftRegister class is designed to have as many outputs as notes you want to respond to.  So a single 4HC595 would have 8 different notes.

If you wanted your single HDD to, for example, play the Low Floor Tom notes (check [this handy map](https://music.stackexchange.com/questions/7227/what-keys-are-associated-to-what-drums-when-connecting-keyboard-to-roland-electr) to see which note numbers correspond to which drums), you would set `FIRST_NOTE=41`, and `NUM_NOTES 8` (since one 4HC595 equals 8 notes).

With those settings, the first output on the 4HC595 would be the Low Floor Tom(41), the second Closed Hi-Hat(42), the third High Floor Tom (43), etc.

DJthefirst, 2020-06-10:
I haven’t played around with shift registers to much but is their and benefit to driving the hdd with a L298n instead of a transistor? The more common way I’ve seen a hdd run as a percussion instrument is to use a simple transistor to pulse the head and then a rubber band around the head to bring it back. This allows a powerful push in the on state and a quick return. As seen [Here](https://youtu.be/5M5ao1D4jWo).

odinokoe4mo, 2020-06-10:
@DJthefirst I think about use transistor but I don't have it at this moment and use L298n. In the future I replace this to transistor.

odinokoe4mo, 2020-06-10:
@Sammy1Am yes, I comment out the lines for FloppyDrives and uncomment the lines for ShiftRegister. L298N controlled like a transistor. I set FIRST_NOTE and NUM_NOTES but it doesn't work again. I have a vague suspicion that my MIDI file is "wrong". Can you if you are not difficult send "right" MIDI file which works?

odinokoe4mo, 2020-06-11:
[Aha — Take on Me.zip](https://github.com/Sammy1Am/Moppy2/files/4763224/Aha.Take.on.Me.zip)
I tried to use this file.

Sammy1Am, 2020-06-11:
Okay, first thing to check: the drums in that file (and by convention in most files) are on channel 10 (so they would, by default, play on the 10th floppy drive, except the Arduino code doesn't support more than 9 devices).  Try using these scripts in the ControlGUI:

- Condition `c==9`
- Sub Address `1`

This should send *only* the drum channel to the Arduino, and send all the notes to the first device connected to the Arduino.

If still nothing is happening, then we're back to note numbers.  It looks like in the MIDI file you included starts with notes 42 and 35 mostly.  So set `FIRST_NOTE=42` instead and see if you get any activity.  You can also change the note script in the ControlGUI to `42` and it will convert *all* drum notes into note 42.

Sammy1Am, 2020-08-12:
Closing for lack of activity.

closed 2020-08-12 by Sammy1Am