samphonic/Moppy2 #102: round robin broken

Montythegreat, 2020-07-06:
johndfoss had this problem, but I have a keyboard connected via a USB to MIDI device, and despite being pre general midi, it works fine on most software. The standard mode on the software works fine with one floppy drive, but when I use the round robin for polyphony, the notes don't reliably start, and when they do, they carry on until I reset the Arduino. It would be super cool if I could fix this, as I think this project is super interesting!

Sammy1Am, 2020-07-06:
Alright, a few parts to this, but I think we can get it figured out:

- How many drives do you have connected?  If it's a number other than 8, have you changed `nextRoundRobinSubAddress(8)` to `nextRoundRobinSubAddress(x)` (where `x` is the number of drive you have) in the ControlGUI?
- Could you be more specific about what "don't reliably start" means?
- The notes hanging (not turning off) sounds like it could be an actual bug, but if you have any more details about that it could help track it down.  Do they *never* turn off?  Is it only the second note that fails to turn off?  Etc.

Montythegreat, 2020-07-08:
I have done the thing with nextRoundRobinSubAddress (I have 2 drives, so I did nextRoundRobinSubAddress(2) ). For me, not reliably start means sometimes they start fine and sometimes they don't start at all. In response to the other question, they hang until they are cut out by another note, which then also last forever. The polyphony just about works, but is even more unreliable than just playing 1 note.

Sammy1Am, 2020-07-09:
Hmm, well there definitely seems to have been a [overlooked line here](https://github.com/Sammy1Am/Moppy2/blob/master/Java/MoppyLib/src/main/java/com/moppy/core/events/mapper/MIDIScriptMapper.java#L159), so that might be the issue.  I don't have any setup to test with at the moment, but I'll try to get this fixed the next time I have things set up.

If you're feeling adventurous, I think the fix is to change:
`nextOpenRoundRobinSubAddress = nextOpenRoundRobinSubAddress%8+1;`
to
`nextOpenRoundRobinSubAddress = (nextOpenRoundRobinSubAddress%numberOfChannels)+1;`

If I'm reading this correctly (and again, I can't actually run the code so it's hard to be 100% sure), note-on events are still getting spread across 8 channels regardless of what argument you passed.  Without a more detailed description, it's still hard to understand exactly how things are behaving for you so I can't tell if this would be causing all your problems or not.  But it certainly seems like that line is wrong.

Montythegreat, 2020-07-09:
Is this in the config file?

Montythegreat, 2020-07-09:
[floppy issue (2).zip](https://github.com/Sammy1Am/Moppy2/files/4895341/floppy.issue.2.zip)
here is a video (the headbangs are due to my stupid skip-find floppy drives, but when it is happening, it shows it is playing in polyphony)

Sammy1Am, 2020-07-09:
It's in the `MIDIScriptMapper.java` file (as linked).  You'd need to make the change and recompile the Java program.

Thanks for the video, they do indeed seem to just never turn off.  Which is puzzling-- I'd expect at least _some_ times that the notes would stop.  I'm not sure I have any other suggestions at the moment, we'll have to see if changing that one line helps, and next time I have a test setup available I can maybe do some more debugging.

Montythegreat, 2020-07-10:
I tried the fix, but unfortiaanately it made no difference. I did it in moppy 2 master, but the GUI is in a separate folder, although I don't think this makes a difference, as moppy 2 master contains no bat file

Sammy1Am, 2020-07-11:
What are you using to build the application?  NetBean?  Command line Gradle?

Montythegreat, 2020-07-12:
I was using notepad to make the edit :)

Montythegreat, 2020-07-12:
I am using the java program that came with it, although I edited the file you said I should

Sammy1Am, 2020-07-12:
Ah, yeah, after you edit the source code, you'd need to recompile the program-- which is a bit of a big step if you're not already doing some programming in Java.

I'm trying to think of other workarounds you could use in the meantime...  You can create a second Mapper in the GUI and configure them so that even-notes go to one drive and odd-notes to the other.  You'd have about a 50/50 chance any pair of notes would work, but it might be enough to get by until I get around to fixing the bug.

Play around with the scripts and see if you get anything working, and tomorrow I'll try to give you exactly what I think will work (not at my computer currently).

Montythegreat, 2020-07-12:
I think I may wait for the bugfix, as I only know simple BASIC and python programming, but thanks for helping anyway!

Sammy1Am, 2020-08-13:
Problem fixed with commit c610146.

It's not fully tested, and it'll be a while before it's officially released, but you're welcome to give it a try by downloading it here:
[MoppyControlGUI-2.1.0.zip](https://github.com/Sammy1Am/Moppy2/files/5066878/MoppyControlGUI-2.1.0.zip)

(It's much larger than the previously release, I think because of some JavaScript engine changes.)

I'm going to close this issue as it's been "fixed", but do let me know if this resolves the problem for you, or if you run into anything else.

closed 2020-08-13 by Sammy1Am

Montythegreat, 2020-08-13:
unfortiaanately, it seems the new version refuses to boot! I am running on windows 10, and it just displays the command prompt for a split second and then nothing happens. I checked the task manager for the java running, but that wasn't there either.

Sammy1Am, 2020-08-13:
Try running the .bat file from an already open command prompt so that any
errors will be displayed (i.e. run `cmd` and then launch the .bat file from
there).  Let me know what error is showing up.

Possibly some sort of Java thing.

Montythegreat, 2020-08-24:
Its working much better now, although the notes hold on forever when playing only one note at once, but act normally when playing two at once, but still, great work! also, notes now play every time you hit a key, rather than only some of the time