samphonic/Moppy2 #103: Question about reset function

tobiaslp12, 2020-07-10:
Hi,

I am trying to alter the code of the floppy drive for use of an EasyDriver.
In the floppy drive code there is this comment:
"Half max because we're stepping directly (no toggle); grab max from index 0"
Can you explain what the first part means?
Also why do you only use the max position of the first array element?
In my case I want to use different scanners and I think they have different max positions. So I should change that, no?
And wouldn't it be better if the reset functions would only move back the amount of the position until 0?
Otherwise there would be loud skipping on a scanner that is already in its default position.

Thank you in advance
Tobias

Sammy1Am, 2020-07-11:
Good questions!

> Can you explain what the first part means?

Sure, some background that will help hopefully first: most recent 3.5" floppy drives support 80 tracks, meaning the read head can move 80 steps forward or backward.  The head moves each time the `STEP` pin on the floppy drive goes from high to low.  If you drop the signal to low too fast for the floppy drive to notice (or too briefly to sufficiently drop the voltage at the end of a longer wire), the step doesn't happen.  So ideally, rather than a short low pulse for each step, you'd want a square wave at the frequency that you want steps to happen.

I accomplished this in code by calling the `tick()` function _twice_ as often as the frequency I want played.  Each call of tick for a particular drive will just toggle the pin (using `togglePin()`) to the opposite of whatever state it's at currently.

This means that if we want to track how many ticks can happen before we want to reverse the drive head, we need to count to 160 ticks even though there are only 80 tracks.  (I used 158 just to add a very small safety margin)  In the `reset()` and `resetAll()` functions, though, each loop in the function calls `digitalWrite()` twice (once high and once low), so we only need to loop 80 times instead of the double 160.

> Also why do you only use the max position of the first array element?

In the `reset()` function, I use the max position of whichever drive is being reset.  In `resetAll()` we need to use the maximum `MAX_POSITION`.  Rather than do the math to calculate the max, I'm just storing it in the 0th element (since it wasn't being used for anything else).

> In my case I want to use different scanners and I think they have different max positions. So I should change that, no?

As noted in the code, "Index zero of [MAX_POSITION] controls the "resetAll" function, and should be the
 same as the largest value in this array," so you'll want to change it to match whatever the largest `MAX_POSITION` value is.  If you decide to reset them using their individual predicted positions (see next answer), you'll probably want to re-write `resetAll` to call `reset` on each drive separately.

> And wouldn't it be better if the reset functions would only move back the amount of the position until 0?
Otherwise there would be loud skipping on a scanner that is already in its default position.

It would 100% be better, yes.  However, there's not (for floppy drives) any way for the Arduino to determine the **actual** position of the read head (might not be entirely true, but Moppy certainly hasn't implemented this functionality).  So if a step is missed, or the Arduino is interrupted, etc., then the `reset()` function might fail to reset the head all the way to zero (or try to go past zero) if it didn't accurately guess where the head was.

For floppy drives anyway, this isn't much of an issue since they (in my experience) can sense when the head has reached zero and won't keep stepping and skipping beyond that.

If you trust that you'll be able to accurately track where scanner is, you can definitely change the code to only reset as many steps as needed.  Alternatively (and more safely), you could try to come up with some sort of cut-off switch when the scanner reaches zero so that it won't ever try to go beyond zero (after resetting the Arduino, or losing power for example).

tobiaslp12, 2020-07-23:
Thanks for your answer. I decided to make a cut-off switch (as you suggested) on both ends so I don't need currentPosition and MAX_POSITION. I also made a version of the reset code so it will reset until the cut-off switch has been reached. I commented it out because I don't need a reset function if every position is the right position and the reset-function took too long to reset.
I might update #53 when I've deleted unnecessary stuff.
I also have a question about rerouting notes through the mapper but I should make a new issue, shouldn't I?

Sammy1Am, 2020-07-24:
Very cool! (would love to see a picture of your switch set up)

And yeah, I think the "correct" thing to do would be a new separate issue
(though things tend to be pretty lax on that in this repo 😉)

tobiaslp12, 2020-07-24:
> Very cool! (would love to see a picture of your switch set up)

Thanks👍. Sure, here is a picture:
https://imgur.com/a/tSkHLrk

closed 2020-07-25 by tobiasfrck

Sammy1Am, 2020-07-28:
Wow, love the switches-- the foam/sponge is a good idea for giving it just a little bit of wiggle room so it doesn't crash into the ends.  Nice.