samphonic/Moppy2 #104: Arduino disappears

Wiedenhoferkilian, 2020-07-24:
Hello Guys,
Sorry for my bad Englisch at first. I am Kilian 14 years old an live in Germany.
I built my own Floppydrive piano using Moppy 2.0 with 4 Floppys because i wait for my other 4 drives.
I wanted to add a hdd connected to an L298N with a second arduino uno. So that we have 2 arduinos all in all.
If i fire up my system my harddrive makes a buzzing noise but that is not the problem. The problem is that whenn i play the Tetris theme the floppys and the harddrive are playing. But if i play something else the arduino from the harddrive just disappears and is not listet anymore in the MoppyGUI. I have already tried an other arduino uno an arduino nano a diverend driver and already burnd a new bootloader and the software of course. I use the normal code on the floppys and the modified code for l298n. 
I hope anyone can help me.
Thanks for your help already.

Sammy1Am, 2020-07-28:
For the second Arduino (the one connected to the HDD), did you change `MoppyCore.cpp` to use the L298N instrument?

If so:
There may be an issue in that code that's causing an issue with the non-Tetris song you're trying to play.  Might need to debug that code.

If not:
How do you have the L298N connected to the Arduino?
Did you change `DEVICE_ADDRESS` in MoppyConfig.h?

(Sorry it took a few days to get back to you, been away from my computer for a bit)

Wiedenhoferkilian, 2020-07-28:
Yes i changed the MoppyCore.cpp to L298N and connected it like it is discribed in your code:
IN1 = 2
IN2 = 3
IN3 = 4
IN4 = 5

and i also changed the DEVICE_ADDRESS to 0x02 because i want to controll the hdd with a diverent mapper.

Sammy1Am, 2020-07-28:
Okay, great, those are all the right things to do. 👍 

What song are you trying to play that makes the HDD-Arduino disappear?  Have you tried "O Cara Mia, Addio (Turret Opera).mid"?

If the Arduino disappears from the list in the GUI, that indicates that it is no longer responding to pings from the control software (which means it's probably crashed or stuck somehow).  There may be something in the MIDI file you are trying to play that causes it to crash, or possibly a bug in the L298N code.

What happens if you try with *only* the L298N Arduino (and unplug the floppy drive Arduino)?

Wiedenhoferkilian, 2020-07-29:
I tried to play the Ghostbusters.mid : http://www.midis101.com/free_midi/56561/Ray_Parker_Jr__Ghostbusters

 . I tried O Cara Mia, Addio.mid and the arduino also disapperars. And it also disappers when i only use the L298N Arduino.

[image]

Sammy1Am, 2020-07-29:
Huh, so it's not something to do with the MIDI file then.

Does the Arduino eventually show up again, or do you have to reset it before it works?
Does the L298N play any notes before it disappears, or does it disappear right away?

It sounds like there's a bug in the L298N code somewhere.  Unfortunately I don't actually have any L298N drivers myself so I'm not able to help debug the code.  I can take a glance through it next time I'm working on the code to see if anything jumps out as an issue, but I won't be able to test any fixes.

Wiedenhoferkilian, 2020-07-29:
No the arduino never shows up again until i uncheck it in "Network Bridges" and check is again. No it does not play any notes but it disapers 10sec after i startet the song.

Sammy1Am, 2020-07-29:
Okay, thanks for the additional information.  I don't have any immediate fixes unfortunately, but I'll try to look at things next time I'm in the code and see if anything stands out.  Sorry.

Wiedenhoferkilian, 2020-07-29:
No problem I think i have found another way to controlle hdds with your software. Maybe can leave this issue open for other users if they have a solution. Thanks for your help.

ShadowBr0ther, 2020-10-27:
Same issue here with EasyDrivers

Wiedenhoferkilian, 2020-10-27:
What are EasyDrivers if i can ask?

ShadowBr0ther, 2020-10-27:
Its a setting ("Instrument type") in Moppy2.

ShadowBr0ther, 2020-10-27:
Try to map your hdd with strict channels: 

Device Ardress: 2
Sub Adress: 1 (not c+1)

Worked for me.

Wiedenhoferkilian, 2020-10-27:
Alright thanks i will try it