samphonic/Moppy2 #106: ControlGui exception prevents startup

tobiasfrck, 2020-07-29:
HI,

my controlgui can't start because I got this exception:
```
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
        at com.moppy.core.events.mapper.MIDIScriptMapper.setNoteScript(MIDIScriptMapper.java:147)
        at com.moppy.control.gui.mapperpanel.MapperPanel.noteComboBoxActionPerformed(MapperPanel.java:493)
        at com.moppy.control.gui.mapperpanel.MapperPanel.access$800(MapperPanel.java:21)
        at com.moppy.control.gui.mapperpanel.MapperPanel$9.actionPerformed(MapperPanel.java:345)
        at javax.swing.JComboBox.fireActionEvent(JComboBox.java:1258)
        at javax.swing.JComboBox.setSelectedItem(JComboBox.java:586)
        at com.moppy.control.gui.mapperpanel.MapperPanel.<init>(MapperPanel.java:66)
        at com.moppy.control.gui.mapperpanel.MapperCollectionPanel.addNewMapper(MapperCollectionPanel.java:60)
        at com.moppy.control.gui.mapperpanel.MapperCollectionPanel.lambda$loadMappersFromConfig$0(MapperCollectionPanel.java:41)
        at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
        at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
        at com.moppy.control.gui.mapperpanel.MapperCollectionPanel.loadMappersFromConfig(MapperCollectionPanel.java:40)
        at com.moppy.control.gui.mapperpanel.MapperCollectionPanel.initMapperCollectionPanel(MapperCollectionPanel.java:31)
        at com.moppy.control.gui.MainWindow.initComponents(MainWindow.java:59)
        at com.moppy.control.gui.MainWindow.<init>(MainWindow.java:37)
        at com.moppy.control.MoppyControlGUI$2.run(MoppyControlGUI.java:79)
        at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```
It has been working until I deleted some programs (other usb drivers e.g.) and deleted some temp files. I reinstalled the arduino ide and the serial monitor is atleast not crashing or something like that.
I downloaded the zip from the releases and it still isn't working.

Does someone know how to fix this?

Sammy1Am, 2020-07-30:
What version of Java are you using?  The Nashorn JavaScript engine is set to be removed with JDK 15, so if you're on the cutting edge that might be the issue.

Otherwise since the error is occurring during initialization of the mapping scripts, my best guess is maybe you deleted some sort of Java temp/preferences storage?  Though preferences *should* be stored in the registry under `HKEY_CURRENT_USER\SOFTWARE\JavaSoft\`, or in `/Users/Alvin/Library/Preferences/` on a Mac.

tobiasfrck, 2020-08-02:
Hi,
I was a bit frustrated with this although the solution was relatively easy. I just uninstalled all java versions I had installed and installed version 14.0.2. Now its working again and I even see these symbols on play, stop and delete mapper.

tobiasfrck, 2020-08-03:
Okay so there is more to this issue than just a "I guess it works now.".
My goal with the scanner was to play high notes that the floppy drives could not handle. Playing notes on its own works great but when I apply a certain mapper everything breaks.
So I want to reroute all notes (this may be the problem) from all channels (I guess this is what "All events" means) to a single channel on the second device which is one scanner. This is my [mapper setup](https://imgur.com/a/5leVonb).
There were already bugs like when I made a change to the note range and chose "Custom" in the note section it resets back to the state before the changes. After a while I looked at the commandline and saw a bunch of exeptions like the one above (not sure if they are exactly alike but they all said "Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException").
I thought maybe thats what the problem was last time. After a quick relaunch of the MoppyControlGui the Gui won't open and there is a slightly different exception from last time which is probably due to the newer version of Java:
```
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
        at com.moppy.core.events.mapper.MIDIScriptMapper.setNoteScript(MIDIScriptMapper.java:147)
        at com.moppy.control.gui.mapperpanel.MapperPanel.noteComboBoxActionPerformed(MapperPanel.java:493)
        at com.moppy.control.gui.mapperpanel.MapperPanel.access$800(MapperPanel.java:21)
        at com.moppy.control.gui.mapperpanel.MapperPanel$9.actionPerformed(MapperPanel.java:345)
        at java.desktop/javax.swing.JComboBox.fireActionEvent(JComboBox.java:1264)
        at java.desktop/javax.swing.JComboBox.setSelectedItem(JComboBox.java:589)
        at com.moppy.control.gui.mapperpanel.MapperPanel.<init>(MapperPanel.java:66)
        at com.moppy.control.gui.mapperpanel.MapperCollectionPanel.addNewMapper(MapperCollectionPanel.java:60)
        at com.moppy.control.gui.mapperpanel.MapperCollectionPanel.lambda$loadMappersFromConfig$0(MapperCollectionPanel.java:41)
        at java.base/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1625)
        at java.base/java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:658)
        at com.moppy.control.gui.mapperpanel.MapperCollectionPanel.loadMappersFromConfig(MapperCollectionPanel.java:40)
        at com.moppy.control.gui.mapperpanel.MapperCollectionPanel.initMapperCollectionPanel(MapperCollectionPanel.java:31)
        at com.moppy.control.gui.MainWindow.initComponents(MainWindow.java:59)
        at com.moppy.control.gui.MainWindow.<init>(MainWindow.java:37)
        at com.moppy.control.MoppyControlGUI$2.run(MoppyControlGUI.java:79)
        at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:316)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
        at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
```
Is this because Moppy tries to send too many notes to the one device or is this unrelated or shouldn't be an issue?
If not what could be the issue?

I reinstalled Java and it did not fix the issue again.

Sammy1Am, 2020-08-03:
Hey there, thanks for the screenshot, that was super helpful in figure out what's going on.

Based on the script you have in the screenshot, the issue appears to be that a script that returns `null` causes a NullPointerException (which is... I guess not super surprising, but failing to start the program is not an intended outcome so I can probably add a check for that, and it **definitely** shouldn't be one of the default available scripts, so I'll remove that, lol).  The issue is that if the script returns `null`, the Mapper will try to figure out how to tell the device to play the MIDI note with number `null`. ☹  So that's where the errors are coming from.

Instead of trying null-i-tize notes that are outside the desired range, it's probably best to put the note-range filter in the Condition script.  So your first example in the screen shot would have a Condition of `n>20 && n<72` (which would return `true` if the note is in that range, or `false` if it's not).  And then your Note script would still just be `n`.

> So the issue seems to be the limits (either max or min) in the note section. The minimum note can't be above 60 (so n>60) and the maximum note can't be below 60 (so n<60). Is this intentional? If so why and if not what is the problem which causes this?

I'm not sure I understand the question, but it might be a bit wrapped up in the issue above.  Hopefully things are clearer now, but if this is still a question, can you clarify where you're seeing the 60 limit?

Sammy1Am, 2020-08-12:
Removed the bad default script, added some checks to look for more script edge cases, and switched from Nashorn to GraalVM (so Java 15+ should be working again).  Fixes will be in version 2.1.0 when it's released (but have been committed already).

Thanks for helping track these down.

closed 2020-08-12 by Sammy1Am