samphonic/Moppy2 #109: How can I make a loop with MoppyClassic?

ysi0430, 2020-09-10:
Hello I'm south Korea highschool student and I've made a moppy with 4 floppys. And I will exhibit it in a few days. But I have a little curiosity "If I got an exhibition, I probably left it alone. But I can't manage because of the COVID-19 so I want to make the moppy to run with loop. Does anyone could help me?
I've made it with UNO
Using Windows 10
If it can, I would modify the .bat file
Do not want to using Netbeans or complicated software

Sammy1Am, 2020-09-10:
Hi there!

Before you try to make that work with MoppyClassic, is there a reason you haven't tried with Moppy 2.0?  It has a lot of improvements, and the latest [2.1.0 update](https://github.com/Sammy1Am/Moppy2/releases/tag/2.1.0) added the ability to create a playlist that plays in a loop (see screenshot below).  You'll need to update the firmware on the Arduino with the 2.1.0 code, but the wiring should be the same, so it should pretty much work out of the box for you.

[image]

If there's some reason you **must** use MoppyClassic:  I'm not sure you'll be able to through a .bat file, because if I remember correctly (it's been a *long* time since I touched MoppyClassic) there's no way to auto-play a MIDI file from the .bat file.  You might be able to find some sort of testing software that can emulate mouse-button clicks and record yourself looping the song?  But otherwise I don't think it's possible.

I definitely recommend switching to Moppy 2.0.

ysi0430, 2020-09-10:
> Hi there!
>
> Before you try to make that work with MoppyClassic, is there a reason you haven't tried with Moppy 2.0? It has a lot of improvements, and the latest [2.1.0 [...]

Thanks for replying sir
I have tried moppy 2.1.0 software , but I can't understand how to use it. Could you make a detail for using that software?
(And I need more midi files which composed in lower tone like bass, that can be played by 4 moppys) haha

Sammy1Am, 2020-09-10:
It should be pretty similar to the previous one.  What part are you getting stuck on?  (So I'm not telling you things you already know)

ysi0430, 2020-09-10:
> It should be pretty similar to the previous one. What part are you getting stuck on? (So I'm not telling you things you already know)

I couldn't run the .bat file which enable us to run the moppy
After downgrading to the JDK 8, it doesn't go on the right direction

Sammy1Am, 2020-09-10:
If the .bat isn't running, it's probably a `JAVA_HOME` issue.  Try opening a command prompt, and then running the .bat file from there.  It should show you any errors it runs into.

You'll need at least Java 11 to run Moppy 2.0.  Did you see 8 listed as a requirement somewhere?  I'll need to change that if so.

ysi0430, 2020-09-10:
succeed!!
thank you!!!

Sammy1Am, 2020-09-10:
Awesome!  Glad you got it working!  Hope the exhibition goes well.

closed 2020-09-14 by Sammy1Am