samphonic/Moppy2 #111: Question about Drive Select pins

MrGlockenspiel, 2020-10-09:
Using the drive select pins, is it possible to connect two drives to a single connector? If so, how would I go about wiring that?
Also, if I connect both drive select pins to ground, what would happen?

Sammy1Am, 2020-10-09:
A great question!

The drive select pins were actually originally designed to allow two drives to share a single ribbon cable in computers, so it's in line with their original purpose.  For Moppy though there's no reason you couldn't use more than 2 drives actually*.  

For two drives the step and direction pins would be connected from pins 2&3 on the Arduino to both drives.  Then you could select which drive is active by grounding that drive's drive-select pin.  If you wanted to do this programmatically, you could connect each of the drive's select pins to an Arduino pin, and `digitialWrite(LOW)` the pin to enable to drive, or `HIGH` to disable it.  For two drives, this would still require four pins on the Arduino, so you're not really saving much, but after the first two you'd notice a slight pin-savings (e.g. four drives would only need six pins instead of the traditional eight).

If you connected both drive-select pins to ground, both drives would play the same note simultaneously.  You could potentially use this as an amplitude control (i.e. more or less drives playing to control the volume of that note).

***Caveat:** The current draw from the signaling lines to the floppy drives should be pretty minimal (which is why it's safe to connect the Arduino directly to the lines), but there's a chance that if you connected enough drives in parallel the `STEP` pin on the Arduino might be needing to sink more current than it's designed for.  Two drives should be no problem, but it might not be limitless.  Additionally, more lines/drives might affect the amount of time it takes for the signal to reach a sufficiently low voltage to be counted by the drive, so you may have timing issues with too many drives or too long of wires.

MrGlockenspiel, 2020-10-09:
Ok, thank you so much!

MrGlockenspiel, 2020-10-26:
I am confused about when you said this:
>  four drives would only need six pins instead of the traditional eight
Wouldn't 4 drives still require 8 pins, because it needs 2 direction, 2 step, and 4 drive select pins? Or I am missing something?

Sammy1Am, 2020-10-26:
All four drives could share 1 step and 1 direction pin, plus 4 drive select pins.  This *would* mean that either all 4 drives would need to be playing the same note though (and you could add more or less drives based on the select pins), so that might have been my misunderstanding of what you were describing.

You could maybe do some clever drive-select switching to try to interleave the notes so they could all play different notes, but I'm not sure how practical that would be.

Sammy1Am, 2020-11-06:
Closing to keep things clean, but reopen if you have any additional questions

closed 2020-11-06 by Sammy1Am