samphonic/Moppy2 #112: scanner to L298N

zseawq123452, 2020-10-27:
I connected the scanner and the L298N
However, when executed, the scanner motor only makes a small sound, and the header does not move. Which one more need to be modified?

Sammy1Am, 2020-10-27:
I don't have any experience with L298Ns, so I can't help too much.  Can you manually trigger the L298N, and does it work as expected then?  (I.e. is the issue on the L298N-side of things, or the Arduino side of things)

zseawq123452, 2020-10-29:
Yes, I solved the low sound of the motor. But still, the header doesn't move. Is there anything I can modify in the code?

Sammy1Am, 2020-10-30:
You'll need to figure out what the issue is (the head not moving is just a symptom) before we can really talk about modifications.

Are you able to manually trigger the L298N (using the signal lines you've hooked to the Arduino) and get it to move?

zseawq123452, 2020-11-03:
Yes I triggered it manually and it moved.

Sammy1Am, 2020-11-06:
Alright, it sounds like the connection from the L298N to the scanner is fine then.  My only additional troubleshooting tips are:
- Confirm that you *commented* the `#define INSTRUMENT_FLOPPIES` line, and you that you *uncommented* the `#define INSTRUMENT_L298N` line in MoppyConfig.h.
- Double check that there are no errors in the Arduino IDE when you upload the firmware (sometimes it's very subtle about its failures)
- Double check the connections to the Arduino.  If there's any ambiguity, maybe try a different pin?  (I've never connected an L298N so I don't really know anything about the pinout)

closed 2021-01-06 by zseawq123452