samphonic/Moppy2 #113: L298N 12th channel device plays 11th's channels melody

nerogamecraft, 2020-11-06:
Hello. 
One Arduino Uno in floppy mode plays 8 drives without issue, but the second Uno behaves strangely in L298N mode. 
The last HDD connected to the analog ports does not play its 12th line, but repeats the 11th one.
 I've tried swapping L298N wires, but always the same problem. 
Is this the case in the analog ports of Arduino?
 Connection diagram and mapper settings: https://drive.google.com/file/d/114D718Q_6Gq2fWYJdtejj2qt0Svbgoc0/view

Sammy1Am, 2020-11-06:
Thank you for the diagram; I've never used L298Ns so that's helpful in debugging.  It looks like there's a small bug in the code on [these lines here](https://github.com/Sammy1Am/Moppy2/blob/e6ab14abf8b5583a01ae19d5a1b54c1c3280e2e3/Microcontroller/Moppy2-Arduino/src/MoppyInstruments/L298N.cpp#L174-L181).

Inside that code block, try changing all the `[3]`s to `[4]`s.  It should look like:
``` c
  if (currentPeriod[4]>0){
    currentTick[4]++;
    if (currentTick[4] >= currentPeriod[4]){
      step(4,14,15,16,17);
      currentTick[4]=0;
    }
  }
}
```

Reupload the Arduino code and try again.

Let me know if this works and I'll try to make a quick edit to the code.

nerogamecraft, 2020-11-11:
Work fine now! Thanks!

closed 2020-11-11 by nerogamecraft