samphonic/Moppy2 #115: Use of HDDS

Paulo-Cesar-Br, 2020-12-04:
Hello, I am from Brazil so please disregard any English errors as I am using Google Translate. So, I'm using Moppy to play some stepper motors and floppy drives, but I wanted to add my drums to my orchestra with the HDDS hitting the reading head. I don't know if it is necessary to make any modifications to the project or if it is already compatible with that, and if it is how can I use it? How should I connect the drives on the arduino? and whether anything else would be needed to make it work. Thank you very much in advance.

Paulo

Amakes, 2020-12-07:
I am also trying to figure this out.

Paulo-Cesar-Br, 2020-12-07:
@Sammy1Am

Paulo-Cesar-Br, 2020-12-08:
@Sammy1Am

Sammy1Am, 2020-12-09:
No need to repeatedly tag me, I see all the issues as they get created.  A general tip for GitHub: it's often worth your time to [search previous issues](https://github.com/Sammy1Am/Moppy2/issues?q=is%3Aissue+hdd) to see if anyone else has asked the same question.

As you'll see there, HDDs aren't immediately supported by Moppy out of the box, but can be connected without too much additional work.  The biggest difference is that instead of the Arduino sending a square wave at a specified frequency, you'll just want a single pulse with the `note on` event.  An example of this can be found in the [ShiftRegister instrument](https://github.com/Sammy1Am/Moppy2/blob/master/Microcontroller/Moppy2-Arduino/src/MoppyInstruments/ShiftRegister.cpp), though you'll need something slightly different for HDDs.

closed 2020-12-09 by Sammy1Am