samphonic/Moppy2 #117: Moppy 2.10 won't open

Amakes, 2020-12-07:
When I try to open Moppy 2.10 it opens a file that looks like a terminal, and then it closes it after a second.

Sammy1Am, 2020-12-07:
Probably a JAVA_HOME or Java versioning issue.  You can check the log file (somewhere near the `.bat` file you're opening), or you can run the `.bat` file from the command line instead (Open up the command prompt first and then type the name of the `.bat` file) to see the error.

You'll need to have JAVA_HOME set properly, and have at least Java 12.

Amakes, 2020-12-07:
How would I fix the java issue?  It is the newest version of java because I installed it yesterday.

Sammy1Am, 2020-12-07:
What error are you getting?

Amakes, 2020-12-07:
I says it is not a internal or external .bat file that it can open

Amakes, 2020-12-07:
Here is more precisely what is says:
'This' is not recognized as an internal or external command,
operable program or batch file.

Sammy1Am, 2020-12-07:
That just means it can't find the .bat file you typed in.  Make sure you're
in the same directory as the .bat file before running it, and make sure
you've spelled the file name correctly.

Amakes, 2020-12-07:
This is what it is saying this time:
Error: A JNI error has occurred, please check your installation and try again
Exception in thread "main" java.lang.UnsupportedClassVersionError: com/moppy/control/MoppyControlGUI has been compiled by a more recent version of the Java Runtime (class file version 55.0), this version of the Java Runtime only recognizes class file versions up to 52.0
        at java.lang.ClassLoader.defineClass1(Native Method)
        at java.lang.ClassLoader.defineClass(Unknown Source)
        at java.security.SecureClassLoader.defineClass(Unknown Source)
        at java.net.URLClassLoader.defineClass(Unknown Source)
        at java.net.URLClassLoader.access$100(Unknown Source)
        at java.net.URLClassLoader$1.run(Unknown Source)
        at java.net.URLClassLoader$1.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.net.URLClassLoader.findClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.launcher.LauncherHelper.checkAndLoadMain(Unknown Source)

Sammy1Am, 2020-12-07:
Version 52 indicates that you're trying to run Moppy using Java 8.  If
you've installed multiple versions of Java, you can try removing the older
versions, but you may need to adjust `JAVA_HOME` to point to the version
you want to use.

Amakes, 2020-12-07:
How would I adjust JAVA_HOME

Amakes, 2020-12-07:
I just got it to open! Thank you for all your help. Apparently it had installed version 8 again.
Thanks Again.

closed 2020-12-07 by Amakes

Sammy1Am, 2020-12-07:
Glad you got it working!