samphonic/Moppy2 #122: How to edit a midi file?

shojkeee, 2021-01-12:
Hello. I have a question. There are midi files. I have 8 floppy disks. And half of the notes are not played. How do I edit a midi file so that it works correctly?

[image]

Here, for example, is a table of notes. How do I parse a Midi file to these notes?

Sammy1Am, 2021-01-22:
Well, you have a couple options:

- If you're looking to edit the MIDI file, there are a lot of programs that will help you do that, though I found that [AnvilStudio ](https://www.anvilstudio.com/) is a very nice, free program to get started with.  You'll need to make sure that all the notes are between C0 and B4.
- You can also do this programmatically in the GUI using the `Note` script.  There's a template there called "Force into Range" that will lower any notes outside the range by wrapping them around to the lowest note again, but this may not be the ideal solution for your MIDI files.  Generally you'll just want to make sure any notes outside 12-71 get pushed back into that range.

shojkeee, 2021-01-22:
thanks. But I was interested in exactly the channels of the breakdown. I used the program MidiEditor and could not split one channel into several. I'll try the program that you recommended.

shojkeee, 2021-01-24:
it is one channel that can be divided into several channels. What is the best way to do this?

Sammy1Am, 2021-01-25:
Just to clarify: is there only one __channel__ or one __track__?  If
there's multiple tracks all assigned to the same channel, you can just
reassign them to different channels using e.g. Anvil Studio.

Unfortunately if the MIDI file is all in one track things are more
challenging.  Some options:
* Manually reorganize the notes.  This will be time consuming, but would
probably produce the best output.
* Use the Condition mapping script in the MoppyControlGUI to select only a
range of notes for each drive.  E.g. `n>=12&&n<24`.  You'll have to add a
separate mapper for each drive, and manually hard code the sub-address
since the channel is the same for all of them.
* You can use the Round Robin mapping option to just pass that one channel
to all the drives.  The results won't be great, but it's straightforward.
* I know this is probably a bit of a non-starter, but most MIDI files have
separate tracks you can assign to separate channels.  Maybe start with a
different song.

shojkeee, 2021-02-09:
What does this command mean? `n>=12&&n<24` . You don't have any documentation? who I could get to know and ask my commands to?

Sammy1Am, 2021-02-09:
Ah, yeah, things are fairly lightly documented.  The mapping scripts have four variables pre-defined (you can see the implementation [here](https://github.com/Sammy1Am/Moppy2/blob/master/Java/MoppyLib/src/main/java/com/moppy/core/events/mapper/MIDIScriptMapper.java)):
`c` - The zero-indexed MIDI channel
`n` - The MIDI note number
`v` - The velocity value
`midiCommand` - The MIDI command from the message

These scripts are JavaScript that is evaluated by the GraalVM, so they're not specific commands, per se.  The condition script (the first one) is evaluated as a boolean to determine if that MIDI event should be processed at all.  The other three scripts return numeric values representing the Device Address, Sub-Address, and Note to be played, respectively.

So in the case of  `n>=12&&n<24`, it will only return `true` if the note being played is between 12 and 24.  Any notes outside this range will be ignored by that mapper.  You can use this to, as suggested, pass only a certain range of notes to a particular floppy drive.