samphonic/Moppy2 #123: Help wanted: Blank Sketch to start with and Error Compiling for Arduino Mega 2560

GrOS7d8, 2021-01-13:
Hello, I’m currently trying to get the Moppy2 software to work for my Arduino Mega 2560 but I keep running into quite a few roadblocks. To start, I have a blank sketch with comments saying the sketch has been left empty to appease the IDE as well as the top layer code is in the src file in the sketch directory. Also, more functional code are in folders within the src folder storing the main but according to the comments I shouldn’t open them in the IDE and should instead use a text editor or another IDE. At first I simply copied and pasted the code from main.cpp and fixed any directory issues that came up during compiling. Then I kept getting a series of errors leading up to an “Error compiling for Arduino Mega or Mega 2560”. I then tried opening the cpp files on another C++ ide but it turns out the files aren’t executable. I spent all afternoon and almost all night trying to possibly get this to work but to no avail. I do hate to criticize but the tutorial I feel has unfortunately left me lost in the dark. I’ve even done further research to see if I can find any more details I might be missing out in here but so far this is all I was able to find regarding Moppy 2.0. If you could provide further information on what should be done, such as what to do regarding those C++ files in the Moppy 2.0 download, or possibly any further steps that should be taken to set up everything, it would be of great help, as this unfortunately has been quite frustrating to go through. And before the question is asked, yes I have tried uploading the sketch as it was when I first opened it and the IDE said “Access Denied”. Please help, I’m genuinely lost regarding what to do at this point.

GrOS7d8, 2021-01-19:
Here is the entirety of the Error messages:

Arduino: 1.8.12 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\EasyDrivers.cpp: In function 'instruments::EasyDrivers::resetAll()':

sketch\EasyDrivers.cpp:299:27: warning: iteration 2 invokes undefined behavior [-Waggressive-loop-optimizations]

     currentState[stepPin] = LOW;

sketch\EasyDrivers.cpp:297:29: note: within this loop

   for (byte d=FIRST_DRIVER;d<=LAST_DRIVER;d++) {

                            ~^~~~~~~~~~~~~

sketch\src\MoppyInstruments\EasyDrivers.cpp: In function 'instruments::EasyDrivers::resetAll()':

sketch\src\MoppyInstruments\EasyDrivers.cpp:299:27: warning: iteration 2 invokes undefined behavior [-Waggressive-loop-optimizations]

     currentState[stepPin] = LOW;

sketch\src\MoppyInstruments\EasyDrivers.cpp:297:29: note: within this loop

   for (byte d=FIRST_DRIVER;d<=LAST_DRIVER;d++) {

                            ~^~~~~~~~~~~~~

sketch\main.cpp.o (symbol from plugin): In function `instrument':

(.text+0x0): multiple definition of `instrument'

sketch\Moppy2-Arduino.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

sketch\main.cpp.o (symbol from plugin): In function `instrument':

(.text+0x0): multiple definition of `network'

sketch\Moppy2-Arduino.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\main.cpp.o (symbol from plugin): In function `instrument':

(.text+0x0): multiple definition of `setup'

sketch\Moppy2-Arduino.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\main.cpp.o (symbol from plugin): In function `instrument':

(.text+0x0): multiple definition of `loop'

sketch\Moppy2-Arduino.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\src\MoppyInstruments\EasyDrivers.cpp.o (symbol from plugin): In function `instruments::EasyDrivers::dev_noteOn(unsigned char, unsigned char*)':

(.text+0x0): multiple definition of `instruments::EasyDrivers::dev_noteOn(unsigned char, unsigned char*)'

sketch\EasyDrivers.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\src\MoppyInstruments\EasyDrivers.cpp.o (symbol from plugin): In function `instruments::EasyDrivers::dev_noteOn(unsigned char, unsigned char*)':

(.text+0x0): multiple definition of `instruments::EasyDrivers::originalPeriod'

sketch\EasyDrivers.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\src\MoppyInstruments\EasyDrivers.cpp.o (symbol from plugin): In function `instruments::EasyDrivers::dev_noteOn(unsigned char, unsigned char*)':

(.text+0x0): multiple definition of `instruments::EasyDrivers::currentPeriod'

[... 675 lines cut ...]

(.text+0x0): multiple definition of `network'

sketch\Moppy2-Arduino.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\src\main.cpp.o (symbol from plugin): In function `instrument':

(.text+0x0): multiple definition of `setup'

sketch\Moppy2-Arduino.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\src\main.cpp.o (symbol from plugin): In function `instrument':

(.text+0x0): multiple definition of `loop'

sketch\Moppy2-Arduino.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Sammy1Am, 2021-01-19:
Hmm, well it looks like there might be an issue in the East driver code,
but there's a lot of other things going on there too.

What changes did you make to the Arduino sketch after downloading it from
GitHub?  Not sure if it's easy enough to post a diff, but that might be the
most clear.

GrOS7d8, 2021-01-19:
As of this time the only changes made to the code were the directory problems. All other code has been left unchanged.

Sammy1Am, 2021-01-20:
Ah, so, I have a little more context now from your other messages:

There's nothing additional to be done after opening the "empty" sketch.  After downloading from GitHub, just open the `.ino` and hit "upload".  You may need to install the TimerOne library in the Arduino IDE if you don't have it already.  

Even though the `.ino` is empty, the Arduino IDE will find the other code files on its own, compile them, and upload them.  The errors you're getting about things being declared more than once is likely because it's finding the original code *and* the code you've copied into the sketch.

GrOS7d8, 2021-01-20:
I see. I’ll try uploading the “empty” sketch and I’ll update you on this.

GrOS7d8, 2021-01-20:
Good news! The “empty” sketch uploaded successfully!

GrOS7d8, 2021-01-20:
GREAT NEWS! I just tested everything out and opened the bat file and I’m thrilled to report everything is working perfectly! Thank you for your help Sammy!

Sammy1Am, 2021-01-22:
Yay!  🎉 Glad you got it working!

closed 2021-01-22 by Sammy1Am