samphonic/Moppy2 #124: Can't run 2.1.0 GUI (and some about periods)

nerogamecraft, 2021-01-20:
I'm fine with the previous versions, works fine, but the latter only launches an empty console window for a second.
Maybe you need to install some libraries?
Can I use the 2.1.0 sketch with the 2.0.1 GUI?

and what about C9 - B9? Do you have values for them? looks like nema17 will sound good on this notes

nerogamecraft, 2021-01-20:
I also noticed that nema17 does not play notes above F#5, although the values ​​are set for them. I am using floppy instrument in version 2.0.1 and the a4988 and dvr8825 drivers.
Apparently this is a limitation for an adequate sounding of a floppy drive? How can i remove it?

Sammy1Am, 2021-01-20:
Try opening a command prompt first, and using that to launch the `.bat` file, it should display whatever error you're running into (most likely a JAVA_HOME or Java Version error).

In `FloppyDrives.h`, `MAX_FLOPPY_NOTE` defines the highest note that will be played; you can increase that up to 127 to play all MIDI notes.

You can certainly calculate and add C9-B9, just make sure you add them to all three of the arrays in `MoppyInstrument.h` or some functions won't work correctly.  I've been using [this page](https://pages.mtu.edu/~suits/notefreqs.html) as a reference.

closed 2021-01-21 by nerogamecraft