samphonic/Moppy2 #128: Is there a way to change the pinout?

HunterG6700, 2021-02-14:
I am using the ardruino cnc shield with 4 stepper motors so my pins are
step/dir = 2,5
step/dir2 = 3,6
step/dir3 = 4,7
step/dir 4 = 12,13

FPVBadGuy, 2021-02-15:
Check out the file "FloppyDrives.cpp". I think this part is what you are looking for:

  _// Prepare pins (0 and 1 are reserved for Serial communications)
  pinMode(2, OUTPUT); // Step control 1
  pinMode(3, OUTPUT); // Direction 1
  pinMode(4, OUTPUT); // Step control 2
  pinMode(5, OUTPUT); // Direction 2
  pinMode(6, OUTPUT); // Step control 3
  pinMode(7, OUTPUT); // Direction 3
  pinMode(8, OUTPUT); // Step control 4
  pinMode(9, OUTPUT); // Direction 4
  pinMode(10, OUTPUT); // Step control 5
  pinMode(11, OUTPUT); // Direction 5
  pinMode(12, OUTPUT); // Step control 6
  pinMode(13, OUTPUT); // Direction 6
  pinMode(14, OUTPUT); // Step control 7
  pinMode(15, OUTPUT); // Direction 7
  pinMode(16, OUTPUT); // Step control 8
  pinMode(17, OUTPUT); // Direction 8
  pinMode(18, OUTPUT); // Direction 9
  pinMode(19, OUTPUT); // Step control 9_

Sammy1Am, 2021-02-16:
You'll need to start with adjusting pinModes in [`FloppyDrives.cpp`](https://github.com/Sammy1Am/Moppy2/blob/master/Microcontroller/Moppy2-Arduino/src/MoppyInstruments/FloppyDrives.cpp) like @FPVBadGuy points out.

The next important bit is mostly [this line here](https://github.com/Sammy1Am/Moppy2/blob/master/Microcontroller/Moppy2-Arduino/src/MoppyInstruments/FloppyDrives.cpp#L187):
```c 
togglePin(d, d*2, (d*2)+1); // Drive 1 is on pins 2 and 3, etc.
```
There's an assumption there about which pins are assigned to which drives that you'll need to change.  There's probably a few ways to do this, but off-hand a quick dirty way would be to create mapping arrays to lookup the correct pins.  Something like (haven't compiled this, but it should be close):

```c
const int stepPin = {0, 2, 3, 4, 12};
const int dirPin = {0, 5, 6, 7, 13};

// ...

togglePin(d, stepPin[d], dirPin[d]); // This will look up the step and direction pins for each drive
```

There *might* be a performance hit for this, but since we were already doing math there it shouldn't be too bad.  (And in fact because the arrays are constant, the compiler might actually make this faster than the math we were doing before)

HunterG6700, 2021-02-17:
Got it I had to change the list to an array, but that worked,