samphonic/Moppy2 #131: Ramps 1.4

sashastul, 2021-03-16:
I need to use ramps 1.4 board and arduino mega2560
I changed the file code "MoppyDrives.cpp". I activated the new ports and replaced them in function "FloppyDrives:: tick ()"
Now the board does not connect to the program on the computer. What could be the problem?
I attach a file with the changes:
[FloppyDrives.zip](https://github.com/Sammy1Am/Moppy2/files/6147222/FloppyDrives.zip)

Sammy1Am, 2021-03-17:
I'll take a look at the code when I have a chance.  Offhand though, pins 0
and 1 are used (at least on the Uno) for serial communication with the PC.
Are any of the changed pins the serial communication pins?

sashastul, 2021-03-17:
No, the mega2560 pins 0 and 1 are also used for serial communication, but the problem occurs when using pins 54 and 55

Sammy1Am, 2021-03-17:
I don't know if it's the *only* issue, but at least in `togglePin` the `pin` and `direction_pin` arguments are used as indexes for the `currentState` and `currentPosition` arrays.  Your pin numbers will end up out of bounds for those arrays.  You may need to increase the array size either way, but you can either do some math to shift the pins a bit so it doesn't have to be quite so big (e.g. `currentState[pin-24]`).