samphonic/Moppy2 #132: change channel

MineJOT1, 2021-03-21:
Hi, how do change drive/motor channel to like channels 1-8 to channels 9-16?

Sammy1Am, 2021-03-21:
I'm not sure I understand exactly what you're asking.

You can edit the MIDI file to change the channel of a particular set of
notes.  You can use the sub-address script in the GUI to map existing
channels to other sub-adresses on the Arduino (e.g. instead of `c+1` you
could use `c+9`).  And you can change which sub-adresses a particular
microcontroller is listening for by editing the Arduino code, but except in
specific circumstances this is less likely to be the right choice.

Probably the second option is what you want, but with a little more detail
about what you're doing, I could maybe provide clearer help.

MineJOT1, 2021-03-22:
I mean I want use one arduino play channels 1-8 and other to play channels 9-16. I got the Code work 6 months ago but i lost it.

Sammy1Am, 2021-03-22:
Ah, okay.  Three steps:

1. For the second Arduino, change the `DEVICE_ADDRESS` in [MoppyConfig.h](https://github.com/Sammy1Am/Moppy2/blob/f29bb48662e88dfcc02dcfe56122ab190d12693e/Microcontroller/Moppy2-Arduino/src/MoppyConfig.h#L36) to `0x02`
2. Limit the first mapper in the GUI (see screenshot) to only listen for channels 1-8.
3. Add a second mapper that listens to 9-16, assign it to device 2, and subtract 7 from the channel value (channels are confusingly zero-indexed)

That should do it.
[image]

MineJOT1, 2021-03-22:
It worked! Thanks! You maybe put this tutorial to next update?

closed 2021-03-22 by MineJOT1