samphonic/Moppy2 #134: Error when compiling

GMRGaming, 2021-04-01:
When trying to compile the code to my Arduino, I got this error:

Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Arduino Uno"

D:\Moppy\Moppy2-Arduino__1_\Moppy2-Arduino__1_.ino: In function 'void setup()':

Moppy2-Arduino__1_:100:5: error: 'instrument' was not declared in this scope

     instrument->setup();

     ^~~~~~~~~~

D:\Moppy\Moppy2-Arduino__1_\Moppy2-Arduino__1_.ino:100:5: note: suggested alternative: 'strnlen'

     instrument->setup();

     ^~~~~~~~~~

     strnlen

Moppy2-Arduino__1_:103:5: error: 'network' was not declared in this scope

     network.begin();

     ^~~~~~~

D:\Moppy\Moppy2-Arduino__1_\Moppy2-Arduino__1_.ino: In function 'void loop()':

Moppy2-Arduino__1_:111:5: error: 'network' was not declared in this scope

     network.readMessages();

     ^~~~~~~

exit status 1

'instrument' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Any help would be greatly appreciated as I am VERY new to Arduino.

Sammy1Am, 2021-04-01:
Hmm, that is a bit unusual.  How are you downloading / opening the Arduino code?

GMRGaming, 2021-04-01:
Well I downloaded the code and unzipped it to my D Drive, opened the Moppy2-Arduino ino file, and pasted the "main" cpp file into it. This could have been completely wrong and you might be cringing, but as I said, I'm very new to Arduino.

GMRGaming, 2021-04-01:
Forgot to mention I did install the Libraries as well

Sammy1Am, 2021-04-01:
Heh, no cringing, just a bit of chuckle 😉.  I'm still figuring out c(++) myself, so no worries.

There's no pasting necessary; just open the `.ino` file exactly as downloaded and the Arduino IDE will magically get all the code from `main.cpp`.  This is weird and confusing, but for some reason that's how they've decided the Arduino IDE should work.

GMRGaming, 2021-04-02:
Well it did work and I heard the tone once, but didn't hear it upon resetting so I tried to reupload the code and now I have this error Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Arduino Uno"

sketch\src\MoppyInstruments\EasyDrivers.cpp: In function 'instruments::EasyDrivers::resetAll()':

sketch\src\MoppyInstruments\EasyDrivers.cpp:299:27: warning: iteration 2 invokes undefined behavior [-Waggressive-loop-optimizations]

     currentState[stepPin] = LOW;

sketch\src\MoppyInstruments\EasyDrivers.cpp:297:29: note: within this loop

   for (byte d=FIRST_DRIVER;d<=LAST_DRIVER;d++) {

                            ~^~~~~~~~~~~~~

Sketch uses 7082 bytes (21%) of program storage space. Maximum is 32256 bytes.

Global variables use 931 bytes (45%) of dynamic memory, leaving 1117 bytes for local variables. Maximum is 2048 bytes.

An error occurred while uploading the sketch

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x05

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x05

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x05

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x05

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x05

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x05

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x05

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x05

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x05

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x05

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Sammy1Am, 2021-04-02:
The bit about EasyDrivers is a "warning" rather than an error, so it won't stop compilation or uploading.

The avrdude errors are not specific to Moppy, so Google will be a more complete resource than I will.  It sounds like your computer can't communicate with the Arduino though.  Try unplugging it from the USB and plugging it back in.

What sort of power supply are you using for the floppy drives?  And are you using the same power supply for the Arduino?

GMRGaming, 2021-04-02:
Okay. Final questions and then I'll leave you alone lol. I haven't figured out how to download or open the GUI yet for loading MIDI files with and I also want to eventually use all 6 drives that I have but don't know how to go about it.

Sammy1Am, 2021-04-02:
Have you checked out [Getting Started](https://github.com/Sammy1Am/Moppy2/wiki/Getting-Started) on the wiki?  Should point you in the right direction.  But, tl;dr:

1. Download from here: https://github.com/Sammy1Am/Moppy2/releases
2. Unzip
3. Open by double-clicking
4. Drive 1->Pins 2&3, Drive 2->Pins 4&5, Drive 3->Pins 6&7, etc.

🙂

GMRGaming, 2021-04-02:
When I open it, it only shows a bunch of folders. I see the Jar called Moppy Control GUI but when I open it I get a JNI error. Also any particular placement for ground pins for multiple drive setups?

Sammy1Am, 2021-04-02:
Open the appropriate file in the `bin` folder.  The `.bat` if you're running Windows.

The ground pins all just need to be connected to the same ground reference as the GND pin on the Arduino.  So whether they're all attached to some sort of terminal, or daisy chained, or whatever, will be just fine.

GMRGaming, 2021-04-02:
Clicking the batch file causes the console to pop up for a second then immediately close.

Sammy1Am, 2021-04-02:
Open a commandline first and run it from there; you'll be able to see the
error message that way.  Most likely a mis/un-configured JAVA_HOME or an
out of date version of Java.

GMRGaming, 2021-04-02:
Not sure how to do that. I have the command line open but idk what the code would be. Like I said, not experienced with this stuff.

Sammy1Am, 2021-04-02:
That's fine, but since these aren't Moppy-specific questions, you'll probably be better served getting instructions / answers from elsewhere.  Once you leave Moppy, there's a lot of differences on your particular system and different subjective paths to achieve the same thing.

One potential way to run the .bat from the command line might be:

1. Open the folder containing the .bat in Windows Explorer.
2. Type `cmd` in the location bar of the Windows Explorer window.
3. Type `MoppyControlGUI.bat` in the command prompt and press enter.

If this doesn't work, or you run into issues with those instructions, [Google, as always, is your friend](https://www.google.com/search?q=how+to+run+bat+file+from+command+line&oq=how+to+run+bat+file+from+command+line).

GMRGaming, 2021-04-02:
Luckily it did work after install jdk 11 which was my problem. Now once the midi is imported and com4 (where my arduino is) is selected, I should hear the midi from my drive correct? Because it's silent when I play the midi, but I heard the startup tone once I selected com4.

Sammy1Am, 2021-04-02:
Are you using one of the included sample MIDI files?  If not, try starting with one of those first.

GMRGaming, 2021-04-02:
I figured it out. All is good now. Thanks SO much for the help. I'm sure you noticed from the questions I was asking but this is my first coding-related project. I have wanted to do this for 5 years now and I've finally done it. Your video helped a lot for the hardware side of things by the way.

Sammy1Am, 2021-04-04:
Glad you got it working, and glad you've gotten into the first of hopefully many coding projects!

closed 2021-04-04 by Sammy1Am