samphonic/Moppy2 #135: MoppyInstruments

Mapachedsa, 2021-04-02:
Moppy2-Arduino:25:10: fatal error: MoppyInstruments.h: No such file or directory

 #include "MoppyInstruments.h"

          ^~~~~~~~~~~~~~~~~~~~

compilation terminated.

exit status 1
MoppyInstruments.h: No such file or directory

Sammy1Am, 2021-04-02:
You're going to have to provide more details about what you're doing / expecting to happen.  Did you download the code from the releases page?  Did you unzip the folder?  Which file did you open in the Arduino IDE?  Did you modify any files?  Etc., etc..

ajdenaro80, 2021-05-21:
Arduino IDE
Install the latest Arduino IDE, and install the TimerOne library.
Download the latest release of the Moppy Arduino code, unzip it somewhere convenient, and open Moppy2-Arduino.ino in the Arduino IDE.
If this is your first time, skip to the next step. Otherwise, change any settings in MoppyConfig.h to match your setup. (I have followed all your steps, there is no MoppyConfig.h in the code to even change? it says code /* This file left intentionally blank to appease the Arduino IDE. 
I must be stupid cause I don't get it

ajdenaro80, 2021-05-21:
I also opened some other code to see if it would work and got the same error as the first comment on this thread and I followed all the steps. please help

Sammy1Am, 2021-05-21:
@ajdenaro80, it sounds like this might be your first time using Moppy, so you can go ahead and skip the MoppyConfig.h step for now (but see below* for more information about that).

What's puzzling, now that I'm looking more closely, is that there is no line in the code that contains `MoppyInstruments.h`.  And the error message says it's occurring on line 25 of the `.ino` file, but there are only 23 lines in that file.

Where did you guys get the code you're trying to run?

\* _The `.ino` file is supposed to be empty (that's why it says it's intentionally blank), so that's fine.  Following C(++) convention, the code is in the `src` directory, which is where you'd find `MoppyConfig.h` if you needed to modify it, but again, for your first go you should probably just leave it as is.  The Arduino IDE will pick up `main.cpp` from `src` on its own and everything will work fine even though the `.ino` is blank.  This is weird, but it's to work around some weird Arduino IDE behavior._