samphonic/Moppy2 #136: Multiple Drive Setup

GMRGaming, 2021-04-14:
Hello it's me again lol
In terms of hardware I have everything set up, the drives are in the proper pins and the ground pins are daisy-chained to the ground pin on the Arduino. What I want to know is how to make the Control GUI know that I have multiple drives in order to play more tracks in the MIDI.
Thanks so much for the help and I'm loving Moppy so far.

Sammy1Am, 2021-04-16:
There's nothing that **needs** to be done in the Control GUI for adding additional drives.  If you have more than 8 or 9, you'll need to modify the Arduino code (not sure how many drives you have).

By default the settings in the GUI will assign notes one MIDI channel at a time to the drives.  So channel 1 will be drive 1, &c.

GMRGaming, 2021-04-16:
The amount of drives I have is 6 with only 2 plugged in right now for testing. When I select COM4 which is where my Arduino is I get the startup tone from the first drive and 2 steps from the other one but when I go to play a MIDI only the first one plays music.

Sammy1Am, 2021-04-16:
Only the first drive will normally play startup tones, so that's fine.  What MIDI file are you playing?  Does it have notes on channels 1 and 2?  Try connecting your second drive as if it were one of the other drives (maybe the MIDI file only have notes on channel 3, 4, 5, &c.)

GMRGaming, 2021-04-16:
I was using different MIDI files but to make troubleshooting easier I'm using the Tetris sample song included with the program and I'm still getting the same problem. I swapped out the drive and still got the same issue.

Sammy1Am, 2021-04-16:
Can you provide a screenshot of the GUI just to make sure there's no
configuration issues?

GMRGaming, 2021-04-16:
[image]

Sammy1Am, 2021-04-16:
The `Device Address` script should just be set to `1` since you only have
one Arduino hooked up and its Device Address is `1` (see panel on upper
right).

GMRGaming, 2021-04-16:
Oh I see! I thought that was talking about drives and not Arduinos. That's why I selected one per channel. Thanks again for the help. Now I can hook up the rest, make a table for them to sit on, and I'll be done!

closed 2021-04-16 by GMRGaming