samphonic/Moppy2 #137: need help on moppy 2

Zgarner28, 2021-05-26:
If you're having trouble, try to describe what you've done so far and what *does* work so we don't ask you to try things you've already done.  If you can provide screenshots, logs, or videos, they can really help get a faster response!
well what does work if the floppy drive and the power supply and the microcontroller. but the problem happen when I open the  moppy file in  Arduino I don't get the moppyconfig.h page. I don't know what I'm doing wrong I need as much help as you can give please and thank you
[image]

Sammy1Am, 2021-05-27:
What are you trying to change in `MoppyConfig.h`?  If you're just going for a basic setup, you shouldn't need to change anything.

Zgarner28, 2021-05-30:
[image]

alastaira, 2021-06-16:
> 
> 
> then this pops up when i try to up load it

That's what it's meant to look like. The message tells you where to find the sources files, if that's what you're looking for, but there's no need to change them.

Sammy1Am, 2021-06-19:
Ah, sorry I missed this somehow, but thanks, @alastaira , that's correct.  Everything looks good, you should be able to just press the upload button to upload it to the Arduino.

Zgarner28, 2021-06-19:
I trued up loading it nothing happens I got it working once then never again and the  last time I got working the red code at the bottom of the screen, it wasn't there and actually worked so I'm thinking that the red code is an error

Zgarner28, 2021-06-19:
I feel like I don't know what I'm doing

Zgarner28, 2021-06-19:
It don't help a video of uploading and downloading it to the Arduino and getting it all set up would be nice like a step by step video the one by the person who make the program don't really help because how don't show how to do all the file set up if you know what I'm trying to say @Sammy1Am

Sammy1Am, 2021-06-19:
The red text isn't an "error" per se, it's just a warning; compilation still succeeded, and uploading would likely also be successful.  However, that red text should not be showing up in the files downloaded from the releases page (by default, EasyDriver shouldn't be getting loaded).  Give these steps a shot:

1. Delete the whole Moppy2Microcontroller-2.1.0 folder.
2. Download the release again [from here.](https://github.com/Sammy1Am/Moppy2/releases/tag/2.1.0)
3. Unzip into a folder somewhere.
4. Open Moppy2-Arduino.ino.
5. Click "Upload".

That's it.

There's really only one step in the Arduino IDE which is "press upload"; so I feel like a video of a button being pressed would be uselessly short.  On the other hand, there's a whole mess of issues that you *could* run into using the Arduino IDE; but there's not really a way to make a video covering *all* of them, and most of them are answered on Google anyway.

Zgarner28, 2021-06-19:
@Sammy1Am 
https://user-images.githubusercontent.com/84870564/122656861-094cfb80-d124-11eb-8430-288cc99f2886.mp4

Sammy1Am, 2021-06-19:
Ah, I was mistaken.  My Arduino IDE had compiler warnings set to "None" so I wasn't seeing those warnings.  I see them now, but they're still just warnings and don't interfere with compilation.

What happens when you press the "Upload" button?

Zgarner28, 2021-06-21:
it uploads and i press the button on the uno and it does nothing i know the pinout is right because its the same as i had the last time. i know all the hardware works

Sammy1Am, 2021-07-07:
I'm not sure what else to tell you; the code hasn't changed since you got it working the first time, so something locally must have.

- Have you tried manually connecting the direction pin to ground and tapping the step pin to ground just to confirm the floppy drive and power supply are still working?
- You have the Arduino ground pin connected to a ground pin on the floppy drive, correct?

Zgarner28, 2021-07-21:
Yes

Zgarner28, 2021-07-21:
To be honest I don't think it's working on the software donde on my part

Sammy1Am, 2021-07-22:
Not sure what else to tell you.  It worked once for you, so unless you've made changes on your end, there's no reason I can think of for it to stop working outside of hardware issues.

Likewise, if it worked once and you haven't changed any of the connections, it's not like those would change.  And if you can manually advance the floppy drive head, then the floppy drive is fine.