samphonic/Moppy2 #140: 30 Floppy

MamoNieTeraz, 2021-07-31:
Hello, at the outset, I greet the author, it is a very good job. I'm after connecting the first drive for the first time. I would like to expand my project to 30 disks. Here are my questions. my UNO will handle physically without manifolds 6. Buy an additional 4 UNO or some manifold? What about additional UNOs and the mapping of their stations? Are there any examples that can be modeled on to ensure that the sound is as well separated as possible? Sorry for my english, it is from google uncle. I greet you all warmly :)

Sammy1Am, 2021-07-31:
Hello there, and thank you.  I hope this message translates okay.  If anything is confusing, please feel free to ask again.

Your Uno can actually handle up to 9 drives.  The "analog" pins also act as digital pins.  But this is still not 30 drives.

First, you should know that MIDI files can only have 16 channels.  So with 30 drives, some of the drives may need to play the same notes at the same time.  Or, you may have to get creative about splitting up the 16 channels into 30 different parts.

You have two options for connecting 30 drives:

1. You could get 4 additional Unos (9 drives per Uno would give you up to 36 drives).  You would need to update [these lines in MoppyConfig.h](https://github.com/Sammy1Am/Moppy2/blob/f29bb48662e88dfcc02dcfe56122ab190d12693e/Microcontroller/Moppy2-Arduino/src/MoppyConfig.h#L36-L41) to give each Uno a unique `DEVICE_ADDRESS`, and specify the number of drives with `MAX_SUB_ADDRESS`.  Then in the MoppyControlGUI, you would add 3 additional Mappers on the lower part of the screen, and change the DeviceAddress for each Uno.
2. Alternatively, you could get some bit-shifters (translated as "manifolds" I think) and connect them to your one Uno.  You'll need to comment out `INSTRUMENT_FLOPPIES` and uncomment `INSTRUMENT_SHIFTED_FLOPPIES` [on these lines](https://github.com/Sammy1Am/Moppy2/blob/f29bb48662e88dfcc02dcfe56122ab190d12693e/Microcontroller/Moppy2-Arduino/src/MoppyConfig.h#L16-L19).  And you'll still need to [update `MAX_SUB_ADDRESS`](https://github.com/Sammy1Am/Moppy2/blob/f29bb48662e88dfcc02dcfe56122ab190d12693e/Microcontroller/Moppy2-Arduino/src/MoppyConfig.h#L41) to however many disk drives you have.

I have successfully used method 2 for eight drives, but only on an ESP8266.  I don't know if an Uno's clock speed will be enough to handle the bit-shifters.

Good luck!  Let me know if you have any more questions.

MamoNieTeraz, 2021-08-01:
thank you for the answer :) then, to avoid confusion, it would not be better to do 30 with 32?  do you think I would be able to power and control 4 uno through one active USB hub?  I think I will do it using method 1 :) What is the easy driver instrument for in config ? Regards :)

Sammy1Am, 2021-08-02:
30 or 32 should work just as well.  32 = 2 * 16, so that might help with the MIDI channel limitation of 16 channels.  You could have two drives play each MIDI channel.

I think you could power and control 4 **Unos** with one active USB hub.  **However**, you will need a separate power supply for the 32 floppy drives.  Do not power the drives via the Arduino, the Arduino's voltage regulator would not like that.

The EasyDriver instrument is for controlling stepper motors using the [EasyDriver board](https://www.sparkfun.com/products/12779).  Some people have used this to connect e.g. scanners or other motors.

MamoNieTeraz, 2021-08-02:
Tomorrow (I hope) 32 arduino 3 floppy players and a hub will come to me. I have already managed to get a computer power supply, although I have a problem whether or not to remove the fan from it. I also have a question about analog pins (I know a lot of these questions already)
2>Step
3>Direction
4>Step
.
.
.
.
.
A0>Step
A1>Direction 
A2>Step
I understand well?
Are the analog pins at the beginning of the count or at the end? Tomorrow I hope to start putting everything together, I will definitely send photos when it's ready: D Regards: D

Sammy1Am, 2021-08-03:
You understand correctly.  A0 = 14, A1 = 15, etc.  You can see this [being used on these lines](https://github.com/Sammy1Am/Moppy2/blob/f29bb48662e88dfcc02dcfe56122ab190d12693e/Microcontroller/Moppy2-Arduino/src/MoppyInstruments/FloppyDrives.cpp#L47-L64).

Looking forward to seeing some photos!

MamoNieTeraz, 2021-08-29:
[image]
 finally it worked: D without help it would take a lot more time, thank you very much: D does anyone know any test sounds for all channels? regards

Sammy1Am, 2021-08-29:
Hooray!  Also, that is a very clean setup-- so compact!

I know @Ultrawipf has built a 16 drive setup, so they may have some 16 channel MIDIs they'd be willing to share with you.

MamoNieTeraz, 2021-08-29:
it would be great: D everything is connected but I have the impression that some floppy disks do not play and then they start to play and I do not know myself; p I also noticed that some songs play better 8x2 than 16x1.I am very pleased that we finally managed to put everything together, this project is great: D