samphonic/Moppy2 #145: floppy not playing sound or even startup tone after reset.

joeyjo33, 2021-10-02:
i have followed your instructions from the youtube video but i can't seem to get it working.
my pinout is as following.
pin 11 to 12
pin 18 to pin 3 on the arduino
pin 20 to pin 2 on the arduino
and pin 19 to ground.

Sammy1Am, 2021-10-02:
What happens when you upload the Arduino code?  Is the LED on the drive on?

joeyjo33, 2021-10-03:
nvm got it fixed

joeyjo33, 2021-10-04:
> What happens when you upload the Arduino code? Is the LED on the drive on?
> […](#)

the green led in front is on. and if play the music it works but no startup sound. nice project to do for fun.

joeyjo33, 2021-10-19:
> What did you do to fix it?

i just changed the ground and used a diffrent midi song to see if it
worked. if i used a diffrent set of pins on the arduino. it worked with
other songs