samphonic/Moppy2 #149: Moppy2 controlled by Command Line possible?

jlhirondelle, 2021-11-01:
I'm using Moppy for a group project in university right now, and it involves hooking up several floppy drives to a server, and using Moppy to automatically play music on them whenever a certain type of log is generated by the server. I can't see a good way to do this with the GUI, so I was wondering if it was possible to run Moppy as a command-line sort of thing instead? That way we'd be able to just run a command/set of commands as a task if our thing is triggered, and that would automatically play the MIDI. I'm unsure if this is a thing, but if it already is or you could point us in the right direction, it would be very appreciated

jlhirondelle, 2021-11-01:
On Linux (Ubuntu) by the way, forgot to mention

Sammy1Am, 2021-11-06:
Hey there, sorry it took a bit to get back to you (busy week here!).

There isn't out-of-the-box command line functionality, but there are two (hopefully quick) options for you to look into:

- I know other people have asked about a CLI in the past; it's possible someone has made one somewhere out there on GitHub.  I did a very quick look through the network and Google results and didn't find anything, but you might have better luck if you dig deeper.  Depending on when they implemented it, it might not be compatible with the latest version anymore, but it might get you started.

- I did my very best to make sure all the actual functionality was contained inside the `MoppyLib` library and `MoppyControlGUI` was *only* for controlling and configuration.  Especially if you're just trying to set up something automated and don't need a lot of configuration, it should be pretty easy to write a very light Java wrapper over MoppyLib.

You basically just need [these four lines](https://github.com/Sammy1Am/Moppy2/blob/f29bb48662e88dfcc02dcfe56122ab190d12693e/Java/MoppyControlGUI/src/main/java/com/moppy/control/MoppyControlGUI.java#L43-L46), though you'll need to do some initialization to get the objects you need (some of which might just be do-nothing objects like the statusBus if you're not planning on providing feedback).

If you decide to go that route, let me know if you have any specific questions you run into.  Depending on your timeline, I **might** be able to try slapping something crude together to get you started, but I wouldn't want you to end up waiting on me if I can't get around to it (musical floppies are, sadly, not a life-priority at the moment 😜)