samphonic/Moppy2 #154: I got a problem when trying to run the macrocontrolGUI.bat

khaled040940, 2021-12-11:
So after I finished connecting 1 floppy disk, and the 4 starting notes appeared, I tried to move to the next step which is running the moppycontrolGUI.bat, but here's the problem, when I ran it through cmd (I wrote CMD in the file location bar. and then I wrote "MoppyControlGUI.bat" so it would run it, I get this error, so I tried downloading other version of Java, in this case Java 6. But then it told me the complete opposite and told me to run it in a newer version of Java, so now I'm confused on what to do. I even tried to download the earlier version of Java 8 but same error appeared as shown in the picture below. FYI I'm currently having the older version of Java 8, to be more specific the jdk-8u202.

Another question is, once I run the MoppyControlGUI, what am I expecting and how do I actually play music on it, is it something to do with the lib folder or something? please help me and thanks <3.

[image]

khaled040940, 2021-12-11:
I actually fixed it by downloading java 15, but now I got a problem as in when I wanna check the box of my Arduino which is COM4, it's not checking it and idk why. 
[image]

Sammy1Am, 2021-12-13:
That is a bit weird.

Does the console window stay open after you run the .bat, and do you see any errors printing out in it?  Can you check the UDP box (239...)?