samphonic/Moppy2 #155: Does 74HC595 work with moppy?

MineJOT1, 2022-01-09:
I am thinking about buying solenoids ,shift register and buckets(yes), so would 74HC595 work with moppy?

Sammy1Am, 2022-01-09:
It would.  In fact there's a ShiftFloppies (or similar name, not at a
computer at the moment) instrument already in the code base designed to
work with bit shifters.

Unless you're using very small solenoids though, you may need some sort of
component between the 74HC595 and the solenoids.

MineJOT1, 2022-01-09:
Ah, okay. Yeah, I need to buy some transistors too. Is it possible to use 2 shift registers to use play more drums?

Sammy1Am, 2022-01-09:
The only limitation on the number of shift registers is how long you're
willing to wait to shift all the bits to them (Arduinos aren't super fast,
so too many might start to interfere with other operations).  I've
successfully used 4 of them though without any noticable issues; 2 should
definitely be fine.

MineJOT1, 2022-01-09:
Oh, okay. Then I am gonna buy some components next week and make probably a prototype board because I don't like all the wires and other stuff falling from my table.

closed 2022-01-09 by MineJOT1