samphonic/Moppy2 #157: Floppy drive not playing anything

Aryan3040, 2022-03-09:
I have my pins set up as
pin 11 to 12
pin 18 to pin 3 on the arduino
pin 20 to pin 2 on the arduino
and pin 19 to ground.
the floppy drive is being powered by the Arduino on 5v line
I tried the troubleshooting tactics and the stepper motor moves but the music does not play
The green light is on by the way
I am using the gui version of the software

Sammy1Am, 2022-03-09:
- When you say, "the stepper motor moves," do you mean when you manually tap the wires together?
- Do the startup tones play when the Arduino is reset?
- Does the GUI controller successfully see and connect to the Arduino?  (A screenshot might be helpful for this part)

Aryan3040, 2022-04-14:
yes it moves when i manually tap wires
no it has never played the startup tones
i see data transfer when i execute code

Sammy1Am, 2022-04-14:
If it doesn't play startup tones and doesn't show as a detected device in the GUI, it's likely that for whatever reason the Arduino firmware flashing process was not successful.  Take a close look at the console in the Arduino IDE after you press upload, the error messages can be sort of subtle sometimes.

Aryan3040, 2022-04-18:
Ok, thanks so much