samphonic/Moppy2 #158: Controller Volume and BPM Not Changing

DiamondPrime423, 2022-03-14:
So I have my drives connected to the Arduino and they connect and output sound, but I find that some drives are very quiet depending on where they're plugged in to and some songs play a lot slower than it feels like they should play at (like the Tetris one). I tried changing the volume and BPM in the controller but nothing seems to take effect.  After I change the BPM, once I hit play it changes back to what it was set to before. Are these things that have to be changed in the MIDI itself?

Sammy1Am, 2022-03-14:
Floppy drives only have one volume, so the volume control will have no effect on them.  Playing a drive with or without a disk, or on different surfaces might make them louder (as floppy drives, being as quiet as possible was a feature, so they'll never be *that* loud).

MIDI files usually contain events that set the tempo of a song (and can contain many of them if the tempo changes).  The BPM slider in the GUI will set the MIDI sequencer to play at a new tempo, but subsequent events in the MIDI file will override that setting.  If you'd like to permanently change the tempo of a song, you'll need to edit the MIDI file.

DiamondPrime423, 2022-03-14:
Ah ok, that makes sense. One other question I had was how to connect more than three drives to one Arduino as my board (Arduino UNO) only has 3 ground slots. Is it fine to just attach the ground on other drives to any piece of metal or do I have to get an expander or put more than one in each slot?

Sammy1Am, 2022-03-14:
The grounds all need to be connected to each other, but you can just daisy-chain wires from drive to drive and then connect the last one to the Arduino.  Or connect all the drive ground wires to a single wire that goes to a single Arduino ground.  "Any piece of metal" definitely isn't a good idea though :P ---- On Mon, 14 Mar 2022 15:44:48 -0700  ***@***.***  wrote ----
Ah ok, that makes sense. One other question I had was how to connect more than three drives to one Arduino as my board (Arduino UNO) only has 3 ground slots. Is it fine to just attach the ground on other drives to any piece of metal or do I have to get an expander or put more than one in each slot?

DiamondPrime423, 2022-03-14:
Gotcha, thanks for the help!