samphonic/Moppy2 #160: FDD Not playing musics

ddagebi07, 2022-06-18:
Hardwares:
Arduino Uno
SAMSUNG SFD-321B(FDD)

pin settings
FDD #11 with FDD #12
FDD #18 with Arduino(digital) #3
FDD #20 with Arduino(digital) #2
FDD #19 with Arduino #GND(the one next to digital #13)

Troubleshooting:
PSU: ok
Arduino code: ok
FDD motor: ok(ticks forewards and backwards)

when I send the code to arduino or turn on the MIDI through the GUI Moppy program, it makes little ticking noises and tickes the FDD motor, but it doesn't make any 'music'.
Arduino is powered thorugh the USB cable by my computer, and the PSU is fine.

DJthefirst, 2022-06-18:
Did you change the default pinout in code? I believe the default pinout is pairs of pins alternating step and direction so the first FDD would be pins 2,3 the next 4,5 ect. So I'm not sure how you wired it but switching the step and dir pins could cause clicking.

DJthefirst, 2022-06-18:
For further diagnosis you can test to see if the Arduino is playing music by connecting one of those cheep buzzers between gnd and the Arduino step pin. The buzzer should play the music if it's working correctly. What is the frequency of the clicking is it rapid like the motors stuck or spaced out to every second or so. Also I assume when you said the psi was ok you checked 5v and 12v and it's not a browning out issue.

ddagebi07, 2022-06-21:
> For further diagnosis you can test to see if the Arduino is playing music by connecting one of those cheep buzzers between gnd and the Arduino step pin. The buzzer should play the music if it's [...]

I'm not sure what "cheep buzzer" is, the PSU works fine, and the frequency is about one tick per second.

Sammy1Am, 2022-06-21:
The ticking is a fairly common failure mode for when the `step` and `direction` pins have been swapped (basically the drive will take a "step" every time the direction changes).  Have you tried switching those two pins?

ddagebi07, 2022-06-23:
it worked. Thanks very much!

closed 2022-06-30 by Sammy1Am