samphonic/Moppy2 #162: Using 40 track drives

80486dx4, 2022-07-11:
I am looking to use an old 360k 5.25 floppy drive with this (thing has a nice loud stepper motor) But since the controller is expecting all drives to be 80 track, the 5.25 drive will reach the (final) 40th track then just keep trying to move further without switching direction. Is there some variable or setting that can be changed anywhere to tell the software or hardware to specify one of the drives as having 40 or some other non-standard number of tracks.

Sammy1Am, 2022-07-11:
Indeed there is.  For a 40 track drive, you would set the value in [this array](https://github.com/Sammy1Am/Moppy2/blob/f29bb48662e88dfcc02dcfe56122ab190d12693e/Microcontroller/Moppy2-Arduino/src/MoppyInstruments/FloppyDrives.cpp#L24) corresponding to the 40-track drive to `78`.  There's some additional explanation in the comment above the array, but if the 40-track drive is the drive connected to pins 2 and 3, you'd change the `158` at index 1 to `78`.

If you're *only* using 40-track drives, you can change the value at index 0 as well.

80486dx4, 2022-07-12:
Okay, thanks for letting me know it seems to be working now. Just so I know (since I have some other really old drives with non-standard track counts) Is the conversion rate for track count to array variable: (tracks*2)-2?

And also is there some way to "mirror" the output of 2 drives. What I want is to make a 5.25 and 3.5 both play as though it were unified as drive 1. So if I put a 3.5 drive on pins 2 and 3 and a 5.25 on 4 and 5 would it be possible to get them both to play whatever would be output to drive 1 in a normal configuration (with the only difference being the max position of the 5.25 drive)

Sammy1Am, 2022-07-12:
Yes, or (tracks-1)*2.

Mirroring is definitely possible using the mapping scripts.  When I'm back
at my computer later I'll try to write the script for you, but that's where
to start.

The other option is to just edit the MIDI file and duplicate channel 1 to
channel 2.

80486dx4, 2022-07-12:
Okay, thank you. I don't really know what a mapping script is but I will try to look into it tomorrow. And I have a just encountered another odd issue with the 5.25 drives. After it working when making the modification that you suggested, I then made a single configuration change to the basic config file (switching from 7 to 6 drives) and now the 5.25 (on pins 2 and 3) is colliding with head stops again. I verified that the index 1 was set to 78 and it was, reverting the aforementioned config change did not help either. I even tried to set all indexes (even 0) to 78 and it seems to have done nothing for any of the drives. (all of the 3.5 drives kept going in full range and 5.25 kept colliding with head stops)

If it is at all helpful for figuring out what the issue is: I am using the provided .ino file to do the compile and upload, I used notepad++ to edit all config and .cpp files, and I am on moppy 2.1.0, and using an arduino mega board.

Sammy1Am, 2022-07-18:
Sorry for the delay.

The lower half of the UI has four script boxes that contain snippets of JavaScript that you can use to map notes as they come out of the synthesizer.  The default first row just maps each MIDI channel to a subaddress (1->1, 2->2).  You can add a second mapper to duplicate MIDI channel 1 to device 2 with these four scripts:
`c==0`, `1`, `2`,`n`

If you change the config back from 6 to 7, does the problem go away?  If it does, it seems like it might be an issue with setting the number of drives.  If it doesn't, it's probably coincidence and there's something else broken.  Maybe try redownloading from GitHub, making _just_ the change to 78, and trying again to make sure it's not an inadvertent change somewhere else.

80486dx4, 2022-07-19:
Thanks for the help with the duplicating tracks. I figured it had something to do with those boxes but I was a bit unsure of how to use them.

As for what the issue was: the time it worked I was using Moppy 2.01 and when I re-uploading with the other setting changed I was using version 2.10. I could not ever get it to work in 2.10 but when I reverted back to 2.01 it worked just fine. (Although I didn't see a way to set the minimum track position in 2.01 which was annoying since my 5.25 drive makes a single knocking sound when it hits track 0 so I would have liked to have been able to set min track to 1) Not sure if it could be a bug of sorts in v2.10 or if I am just missing something.

Sammy1Am, 2022-07-20:
Just so we're on the same page: What config change are you trying to make?  And what's your goal for changing it (i.e. what do you expect to be different).

Also, are you saying setting the tracks to 78 doesn't work at all in 2.10, or it only doesn't work when you're making the other config change(s)?

80486dx4, 2022-07-21:
The config change I was first trying to make was setting drive 1 to 40 track and adding a standard 80 track (so increasing the subaddress count by one) 

As far as I have been able to try and test, setting to tracks to 78 has not worked at all for me in 2.10