samphonic/Moppy2 #164: Error compiling for UNO

COMMODOREAMIGA1000, 2022-07-15:
Arduino: 1.8.19 (Windows 10), Board: "Arduino Uno"

C:\Users\[REDACTED FOR PRIVACY]\Desktop\Moppy2-Arduino\src\MoppyInstruments\MoppyTimer.cpp:4:10: fatal error: TimerOne.h: No such file or directory

 #include <TimerOne.h>

          ^~~~~~~~~~~~

compilation terminated.

exit status 1

Error compiling for board Arduino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

WeeMellowToby, 2022-07-16:
you need to add/install timerone.h to the project

COMMODOREAMIGA1000, 2022-07-16:
I'm new to this; could you explain how I do that?

Sammy1Am, 2022-07-18:
Here are instructions for installing Arduino Libraries, just select the TimerOne library and install it:
https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries

closed 2022-07-19 by COMMODOREAMIGA1000