samphonic/Moppy2 #170: java gui program error

philipboxfrmoscow, 2022-10-16:
Hi, i used this Gui on my old pc, but on the new pc, there's linux.
when i tried to execute it, it showed:

Error: LinkageError occurred while loading main class com.moppy.control.MoppyControlGUI
        java.lang.UnsupportedClassVersionError: com/moppy/control/MoppyControlGUI has been compiled by a more recent version of the Java Runtime (class file version 59.0), this version of the Java Runtime only recognizes class file versions up to 55.0

Please help!

Sammy1Am, 2022-10-16:
You need to use a more recent version of Java.  I don't have the version
numbers memorized (and I'm not in front of a computer now), but a quick
Google will tell you what you need.

Oldmanspub, 2023-01-14:
MoppyControlGUI-2.1.0 was compiled with Java15, as it requires class file 59. If you are on old Java version (in your case class file 55 -> java 11), than you need to update your java, or compile source on Java11 to match with your system.

I had issue to run Moppy on Win7 32bit, as it has maximum Java 8, I had recompiled it for Java 8. Attached here. If you have class file issue, you can give a try. I used the same source as Sammy1Am has here, only replaced compile command to implementation in MoppyDeviceGui build.gradle, as my gradle version not supported compile command at dependencies.

[MoppyJava8.zip](https://github.com/Sammy1Am/Moppy2/files/10417328/MoppyJava8.zip)