samphonic/Moppy2 #172: Directly to arduino?

spmurom, 2022-11-07:
Hello. Tell me how to upload a midi file directly to arduino so that you do not need a computer to run

Sammy1Am, 2022-11-07:
I'm not aware of any way to upload MIDI files directly to an Arduino (though it might be possible for small files); you may need to use an SD Card.  But then you'd do something like:

1. Find a library capable of reading and playing back MIDI files ([like this one](https://github.com/MajicDesigns/MD_MIDIFile) maybe).
2. Instead of passing the MIDI messages out of the Arduino, convert them into Moppy-style messages (see Java library).
3. Call appropriate messaging handling function of the existing Moppy Arduino code, but pass in the message you've created rather than reading from Serial.

ihmSelbst, 2024-06-15:
> Hello. Tell me how to upload a midi file directly to arduino so that you do not need a computer to run

@spmurom Did you get it to work? I'm looking for exactly the same.