samphonic/Moppy2 #174: floppy doesn't work after clicking button on arduino :/ Please help

PanJabuszko, 2022-11-15:
I wired all up, but after uploading the code in arduino the floppy does nothing. Help please

Sammy1Am, 2022-11-15:
Does the MoppyControlGUI see the Arudino when you connect it?  (I.e. does it show up in the list of devices on the top right of the UI?)

It sounds like the startup sound isn't playing, which does indicate some sort of problem.  If it does show up in the UI though, can you try playing a MIDI anyway and see what happens?  It's possible the STEP and DIR pins are swapped.

PanJabuszko, 2022-11-17:
Okey, it was good all the time, pins was good. Just in MoppyConfig directory, function that has been playing "intro" was set to OFF. The samplesong that I was trying to play was cara mio addio, and that wasn't playing, other songs works correctry. So... thanks for help, very much :D. 

Can close

Sammy1Am, 2022-11-22:
Glad you got it working!

closed 2022-11-22 by Sammy1Am