samphonic/Moppy2 #176: JRE

MrYoungCS, 2022-11-25:
I know this has been solved before, but could someone point me in the direction of fixing it\

Error: A JNI error has occurred, please check your installation and try again
Exception in thread "main" java.lang.UnsupportedClassVersionError: com/moppy/control/MoppyControlGUI has been compiled by a more recent version of the Java Runtime (class file version 55.0), this version of the Java Runtime only recognizes class file versions up to 52.0
        at java.lang.ClassLoader.defineClass1(Native Method)
        at java.lang.ClassLoader.defineClass(Unknown Source)
        at java.security.SecureClassLoader.defineClass(Unknown Source)
        at java.net.URLClassLoader.defineClass(Unknown Source)
        at java.net.URLClassLoader.access$100(Unknown Source)
        at java.net.URLClassLoader$1.run(Unknown Source)
        at java.net.URLClassLoader$1.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.net.URLClassLoader.findClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.launcher.LauncherHelper.checkAndLoadMain(Unknown Source)

Sammy1Am, 2022-11-25:
You need at least Java 11, but I'd recommend just grabbing the latest
stable version.

MrYoungCS, 2022-11-26:
I thought I had - hadn't realized you had to manually update the java as well by opening the java config tool, selecting view, finding the folder you've installed the JDK to and then disabling version 8! 

Anyone would think adding a button saying 'would you like to make this your default version of JRE' to the install would be helpful!!

Issue seems sorted now - I put these instructions there so anyone else having this issue would be able to sort it quickly.

closed 2022-11-26 by MrYoungCS