samphonic/Moppy2 #177: launching MoppyControlGUI on a mac

mentaluproar, 2022-11-27:
I have Mac an linux stuff, no windows machines.  I haven't touched java since 2004 so I'm having trouble creating a .sh file that replicates what the .bat file does to launch in windows. Has anyone here run moppy on a Mac?

Sammy1Am, 2022-11-27:
Before trying to build your own, have you tried using the shell script included in the release zip?  It's in the same folder as the `.bat` file.

mentaluproar, 2022-11-27:
Exception in thread "main" java.lang.UnsatisfiedLinkError: Can't load library: /var/folders/xm/n92l0yxs5cxfkfxr6d8swb9r0000gn/T/jSerialComm/1669508790816-libjSerialComm.jnilib
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2408)
	at java.base/java.lang.Runtime.load0(Runtime.java:785)
	at java.base/java.lang.System.load(System.java:2011)
	at com.fazecast.jSerialComm.SerialPort.<clinit>(SerialPort.java:248)
	at com.moppy.core.comms.bridge.BridgeSerial.getAvailableSerials(BridgeSerial.java:31)
	at com.moppy.control.NetworkManager.<init>(NetworkManager.java:53)
	at com.moppy.control.MoppyControlGUI.main(MoppyControlGUI.java:41)

I've installed Java 19

Sammy1Am, 2022-11-27:
Alright, I don't have a Linux/Mac system to test on at the moment (I don't think WSL supports GUI programs yet), but the end result of both the .bat and shell script is essentially:
`java --class-path "lib/*" com.moppy.control.MoppyControlGUI`

So if you `cd` to the MoppyControlGUI folder and run that command, it *should* attempt to run the control GUI.  All the extra script stuff is to deal with like, finding Java, dealing with environment variables, etc.  But for a basic run you shouldn't need it.

There's a chance that jSerialComm isn't supported on your system I guess, what kind of Mac do you have?

mentaluproar, 2022-11-28:
14 inch Mac Pro running on apple silicon. I’ll try this command when I get back home.

mentaluproar, 2022-11-29:
bin % java --class-path "lib/*" com.moppy.control.MoppyControlGUI
Error: Could not find or load main class com.moppy.control.MoppyControlGUI
Caused by: java.lang.ClassNotFoundException: com.moppy.control.MoppyControlGUI

Sammy1Am, 2022-11-29:
Ah, okay, seems like it's a jSerialComm thing.  Eventually I'll hopefully get a chance to update to the newest version and then if you're able to test again hopefully it will work.  Seems like maybe an ARM support thing.