samphonic/Moppy2 #178: current state of ESP8266 version

mentaluproar, 2022-11-27:
I see in the documentation there is support for ESP8266 and in the latest version it can be built under the Arduino IDE.  I cant get the Arduino IDE to build for it, as it is missing a dependency I just cannot seem to satisfy, ESPAsyncWebServer.  I have installed ESPAsyncTCP but it's not working so I'm guessing that was the wrong library.  First question: how do I satisfy that dependency?

Second, according to the notes, UDP isn't implemented yet. However I see it in the GUI screenshots.  Is that notation out of date or is it still unfinished and I should use serial instead on an ESP?

Sammy1Am, 2022-11-27:
Apologies for a quick reply, I'm not near a computer or the code so this is
just my best recollection:

- I'm not sure 100% where ESPAsyncWebServer comes from as it's not
explicitly listed.  I only remember building the ESP code in PlatformIO in
VS Code; I think it might come from the Arduino or ESP platform/library.
(Google will probably know more than me here)
- UDP has been implemented and works fine with the ESP8266.  You just have
to swap out the Serial network option in the microcontroller code for the
UDP option.  I think in some Config.h or something?

Sorry, let me know if you're still missing some info and I'll try for a
more detailed explanation when I can get at the code.

mentaluproar, 2022-11-27:
Don't apologize for a quick response.  It's awesome that you provide support for this at all.  I was trying to make my own rig like this with a teensy and a just control the floppy directly as midi instruments, cutting the need for the GUI entirely, but the code was quickly getting unmanageably ugly and my skills are nowhere near yours. Thanks for providing this and providing help using it.

DJthefirst, 2022-11-27:
ESPAsyncWebServer is a library that supports the WebSocket page that lets you auto-config the wifi.  You could try removing the Web socket code if you are ambitious and hard code the Wifi config but when you get the libraries working it's super handy to have the wifi auto config stuff.

For UDP I've had a slight constant delay using the Arduino Libraries on Esp32 of around 1/4 which is not noticeable for one device but coordinating with other esp8266 or with a backing track on a computer becomes messy. Though I haven't tested with Floppy a ton to nail it down. I think it's just the Arduino implementation on an Espressif core. UDP should be a lot faster than what I kept getting. You can always use Serail as a backup and validation before trying UDP.

I also have a platform that just takes raw midi serial on esp32 if you are interested. It is easy to implement a custom serial converter or use something like hairless MidiSerial but it is still missing some parts I am working on. Esp8266 support should be easy to add and I am working on adding Floppy Drives. Already have Steppers, PWM, and Shift registers but I am currently busy writing the GUI in JS for chrome.

mentaluproar, 2022-11-27:
Your comment brings up a good point. How do you coordinate multiple controllers handling multiple instruments on one song?

For example: assume I have 4x 3.5” drives and 2x 5.25” drives. We cannot have them all on the same controller since they are different types of drives right? So we use two controllers. But with 2 controllers, how can we get them to play the same song, each drive playing a different staff? Is there a way to use multiple controllers with one song in the UI?

I did get the UI running mostly in an Ubuntu VM but certain elements aren’t displaying right so I can’t see all it can do yet.

DJthefirst, 2022-11-27:
With Serial the data transfer is fast enough that there is no perceivable difference from the system running the GUI. You transfer data at 9600 baud / 8 is 1200 bytes per second a midi on/off note is 3 bytes which is 400 events per second and little delay bc of the direct connection. 

Moppy has Mapping which allows you to route midi channels to different connected devices. You can edit the song's midi tracks and select which channels to send to each device. For UDP I have noticed a slight delay between the controller and slaves but try it it might work out. LMK if it does. The GUI also allows you to set how the notes are played in groups on each device. Ex round robin on dev-1 Floppys 1,2,3 with Midi ch-1 and play Midi ch-2 on dev-1 Floppy-4 while dev-2 also plays ch-1 on drives 1,2 etc..

The Moppy GUI Handles all the interactions and distribution of notes to Floppies so it is inseparable from the devices without a replacement. I moved this processing on the microcontroller in my program which is why I use esp32s but it allows the instruments to act as regular midi devices without a GUI. Also, I use loop midi to connect my DAW (Mixcraft) to Moppy allowing me to play music and edit it live not just play files. There are a bunch of free Daws to edit files around the internet.

[image]

Sammy1Am, 2022-11-27:
> just control the floppy directly as midi instruments, cutting the need for the GUI entirely, but the code was quickly getting unmanageably ugly

Yeah, been there, lol.  My first attempt was to use a MIDI shield on the Arduino, but it was a mess.  I *might* be able to get it working now that I've got more experience with microcontrollers, but having the GUI is just handy for tweaking stuff.  You can always use the MIDI In option in the GUI to route MIDI straight through if that's easier than MIDI files for some reason (or if you want to play the drives live with a keyboard).

> Also, you should get VS code and the Platform IO extension for VS It is a lifesaver.

I second this if you're going to be writing any code yourself.  The Arduino IDE is fine for just uploading, or the little LED blinking tutorials, but it's a bit limited for anything significant.  VS Code/PlatformIO has a learning curve for sure, but is a good tool to know.

> How do you coordinate multiple controllers handling multiple instruments on one song?

@DJthefirst 's response basically covers it, but the short version is:

- Each controller needs to have a unique `DEVICE_ADDRESS` set [on this line here](https://github.com/Sammy1Am/Moppy2/blob/f29bb48662e88dfcc02dcfe56122ab190d12693e/Microcontroller/Moppy2-Arduino/src/MoppyConfig.h#L36).
- If you plug in multiple controllers via USB, they'll show as different COM ports in the GUI you can select.  For UDP it's multicast so they'll just all be listening.
- Using the mapping scripts in the GUI, send some of the events to each Device Address.  This is commonly done by channel (see second row of mappings in @DJthefirst 's screenshot), but you could do it by note range or something else as well.

mentaluproar, 2022-11-27:
How does the midi in function in GUI work? I have some ideas for playing the instruments through GarageBand and qlab. But those are on Mac and I can’t get the GUI to launch there. I CAN get it running in the Ubuntu VM with some graphical glitches though.

Alternatively, there is a program called Hairless that only works on older Mac OS’s that would enable midi over serial and I do still have a working g4 iMac running tiger. (Such interesting design from apple back then) so that could control this if the GUI dislikes running in a VM. 

BTW, to anyone looking for a board during the parts shortage, the Adafruit metro mini is a drop in replacement for the arduino uno and marginally easier to source.

Sammy1Am, 2022-11-28:
Basically Java asks the OS if there are any MIDI In or Out devices available and populates those two dropdowns.  If you pick a MIDI In device, any MIDI messages received will be piped to the drives in the same way messages from a file would be.  If you wanted to use GarageBand, you'd have to get the MIDI piped into your VM somehow, though there are a lot of virtual cable programs that might be able to do that somehow.

Did you have any luck trying the simplified command I suggested on the other issue to get things working on Mac?