samphonic/Moppy2 #179: ARM64 windows cannot launch MoppyControl

mentaluproar, 2022-11-28:
Error message:
Exception in thread "main" java.lang.UnsatisfiedLinkError: C:\Users\seancarson\AppData\Local\Temp\jSerialComm\1669603000103-jSerialComm.dll: Can't load AMD 64-bit .dll on a ARM 64-bit platform
        at java.base/jdk.internal.loader.NativeLibraries.load(Native Method)
        at java.base/jdk.internal.loader.NativeLibraries$NativeLibraryImpl.open(NativeLibraries.java:388)
        at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:232)
        at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:174)
        at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2389)
        at java.base/java.lang.Runtime.load0(Runtime.java:755)
        at java.base/java.lang.System.load(System.java:1953)
        at com.fazecast.jSerialComm.SerialPort.<clinit>(SerialPort.java:248)
        at com.moppy.core.comms.bridge.BridgeSerial.getAvailableSerials(BridgeSerial.java:31)
        at com.moppy.control.NetworkManager.<init>(NetworkManager.java:53)
        at com.moppy.control.MoppyControlGUI.main(MoppyControlGUI.java:41)

Using openJDK 17

Sammy1Am, 2022-11-28:
Okay, I don't have time to dig all the way into this at the moment, but if you're feeling adventurous:

jSerialComm seems to be having some issues, but I also noticed that was the issue when attempting to launch on your Mac as well, so I think that's our culprit.  Here's at least two things to try (in either order):

- Update the MoppyLib [build.gradle](https://github.com/Sammy1Am/Moppy2/blob/f29bb48662e88dfcc02dcfe56122ab190d12693e/Java/MoppyLib/build.gradle#L22) to use `2.9.+` instead of `2.6.+`.  And then you'll need to rebuild MoppyControlGUI using Gradle to properly pull down the dependencies.
- [jSerialComm's troubleshooting section](https://github.com/Fazecast/jSerialComm#troubleshooting) mentions setting Java's `os.arch_full` system property to specify your architecture in cases where jSerialComm may have issues detecting it.  They specifically call this out for ARM systems, so this might help (though if version 2.6 doesn't support ARM yet, you'll still need to update to 2.9.

If you do manage to get either of these to work, let me know.  Otherwise this might need to wait until I have time to setup a dev environment again.

mentaluproar, 2022-11-28:
-blink-

huh?

I’ll have to wait for you then. Obviously there’s no hurry. I appreciate how responsive you’ve been.