samphonic/Moppy2 #180: Only 2 drives work on imported MIDI

mentaluproar, 2022-12-01:
With the included MIDI, I can get 3 drives working, round robin gets 4 working (with pauses since it supports 8 and I haven't hooked those up yet). But when I throw some of my stuff at it, it wont play any more than the first 2 drives.  I'm guessing there's something wrong with my MIDI files but when I throw them into GarageBand, all the instruments are there. Is there a known issue with exported midi from flat.io?

mentaluproar, 2022-12-01:
https://mega.nz/file/NERDlaJL#iaA1rHnbd2SID9GsZb_DU0bo1zc0ITk1t_CjEU8xbX4

Sammy1Am, 2022-12-02:
They are not (which is good news, because I think that's the source of the issue).  As you can see in the image below, there are only two MIDI channels in that file (channel 1 and 2).  Channel 1 (the top staff) contains instances where multiple notes are being played at the same time (i.e. chords), but because each channel is being mapped to a single floppy drive, only one of those notes will play.

It does look like there is a second channel for the electric bass though, so those notes would be sent to the second drive (hence two drives working).  You'll need to split the first channel into at least two channels for these to sound correct (there's a couple instances of 3-note chords, but they look like maybe an error (? not a music major, haha) so probably 2 will be fine.

[image]

mentaluproar, 2022-12-02:
Which file did you look at? I think flat is collapsing my staffs on export.

Sammy1Am, 2022-12-02:
They all looked like that I think, or at least 2 of the 3.