samphonic/Moppy2 #183: MoppyControlGui Doesn't work

Ember23, 2023-02-15:
I do not know what am I doing wrong. I installed the program to my arduino UNO properly and when I'm about to open up the control GUI it just doesn't work. A window pops up but then it instantly closes. I am not an expert, so it is possible that I missed out something but I'm not sure. I have a windows 10 (64 bit) machine with 8 gb of RAM and an Intel core I5 porcessor. Please someone help me.

Sammy1Am, 2023-02-15:
If you launch the .bat from the command line (instead of double clicking)
it should show you any errors it's encountering.

My guess is it's a Java version issue; make sure you have the latest
version of Java installed.

thiagooooo00p, 2024-03-02:
how do i load it from command line?

Sammy1Am, 2024-03-03:
Open Terminal (or cmd.exe) in the `bin` directory containing `MoppyControlGUI.bat`, and type `MoppyControlGUI.bat`.

surgood186, 2026-02-07:
how do I do this? I'm just a beginner