samphonic/Moppy2 #184: First drive stops working when i connect second one

T0biasCZe, 2023-03-01:
I have two floppy drives, both work when standalone, but when i connect the second drive while 1st is connected, the first one stops playing correctly and just starts skipping at the start
[floppy.webm](https://user-images.githubusercontent.com/44525446/222220908-03ea1241-8446-45da-9e7a-84c3ef870242.webm)

Sammy1Am, 2023-03-01:
Hmmm, that's an interesting one.  Thanks for the video though that helps a bit.  A couple quick things to check:

- How are you powering the drives?  I think I see a separate power supply, but just making sure they're not being powered through the Arduino.
- Do you have at least one odd pin from each of the floppy drives connected to the Arduino's ground pin?

There's no "detection" of what's connected to the Arduino, the firmware will just keep doing the same thing regardless of what's plugged in, so this points to some sort of wiring/hardware/floppydrive weirdness.

T0biasCZe, 2023-03-01:
they are connected through old 200W ATX power supply, and there are few resistors between 5V and gnd to create load for the psu
yes i have the ground pins on each floppy data pins connected to arduino gnd (though it would be enough to connect just 1 floppy, since the data ground and power ground is connected... so it would go from data pin on floppy 1 to power pin on floppy 1, then over the atx power cable onto the second floppy power and then to second floppy data.... :DDD)
also, it odes like this only when floppy 2 is playing. if floppy 2 is not playing any sound, then the floppy 1 starts acting normally

Sammy1Am, 2023-03-02:
> (though it would be enough to connect just 1 floppy, since the data ground and power ground is connected... so it would go from data pin on floppy 1 to power pin on floppy 1, then over the atx power cable onto the second floppy power and then to second floppy data.... :DDD

In **theory** this is true, but consequently floppy ribbon cables would really only need half as many pins (why put them in there at all if the computer's PSU is grounding both the drive and the motherboard?).  In practice, things can be a little more nuanced, so it's worth checking :).

- Just out of curiosity, do things work if you remove the resistors between 5v and ground on the power supply?
- If you switch the drive assignments (i.e. use Drive 1 as the second drive, and Drive 2 as the first drive), is the same _physical_ drive the one that "wins" if they're both playing, or is it based on the Arduino pins?

It sorta seems like there are some signaling issues between the Arduino and the drives-- weird that it's always the 1st drive that drops off.  I'm asking about the ground a lot because the drives "sense" the data coming from the Arduino when the voltage drops below a certain point, so if the ground is noisy or fluctuating or something, it might have trouble receiving all the data.